FPGAwars/icestudio #554: nextpnr "Target frequency" always 12.00MHz irrespective of the actual FPGA Clock frequency used on selected Board

John-Westlake, 2021-11-20:
Supported iCE40 boards such as Alchitry CU iCE40-HX8K and iCE40HX8K-EVB feature a 100MHz FPGA clock, however it appears that iCEStudio does not recognize the selected Boards actual clock frequency - and defaults to 12MHz.

When nextpnr is invoked by iCEStudio its target frequency is set to a "default" of 12MHz - resulting in timing issues when using higher speed FPGA clocks (such as the 100MHz fitted as standard on Alchitry CU iCE40-HX8K and iCE40HX8K-EVB boards etc).

After little digging, the it appears a default nextpnr Clock Constraint can be applied by --freq {freq} (in MHz), however I have no idea how to achieve this when iCEStudio invokes nextpnr (I cannot modify the nextpnr "command line").

I really would appreciate some urgent help please, as I currently cannot complete my project with the timing issues as a result of the default "too low" 12MHz clock - is there an easy method (under Windows8 / iCEStudio version 0.7.1w202110210410) to set a 100MHz FPGA clock?

Ultimately, it be great to have an option in iCEStudio under say the "Select" tab to choose Default or user selectable FPGA frequency - for those who have modify there iCE40 FPGA boards or whose Boards have higher clock frequency's then the default 12MHz.

cavearr, 2021-11-23:
Hi John! i'm reviewing your issue and i'm trying to help you.

First , it's important understand how its works.

Icestudio don't configure the base frequency clock, it takes it from an external clock from FPGA pin.

In this way your problem is not of Icestudio, the problem resides in the block collection that uses 12Mhz as base clock (hardcoded) to calculate differente freqs,etc.

What collections are you using? i have and idea to try to convert automatically current collection from 12Mhz to other base clock.

Tell me what collections are you using, and i'm trying to  migrate it.

Thanks!

John-Westlake, 2021-11-23:
Hi Caverarr,

Thank you for making the time to look into this for me.

I'm currently using a iceFUN PCB modified with its the default FPGA clock (12MHz) replaced with a 100MHz clock. So I'm aware that iCEStudio does not determine the actual FPGA hardware clock speed.

However, lets say I build a project using the Alchitry CU iCE40-HX8K (or the iCE40HX8K-EVB) these FPGA boards by default have a 100MHz clock.

The issue is that the "**_nextpnr_** timing driven FPGA place and route tool" needs to be made aware of the FPGA "Target Frequency" clock constraint so that that is can try and route the FPGA to the required timing "Budget" (this "Clock Constraint" is configured via the command line used to invoke nextpnr)...

When "nextpnr" command line is invoked by iCEStudio - it appears that iCEStudio applies a default frequency of 12MHz for all Boards - irrespective of the boards actual FPGA operating frequency.

I've attached a snap shot of the nextpnr Clock Constraint command line info below:

[image]

[image]

Many FPGA boards allow external FPGA clocks (apart from the onboard default clock) so iceStudio (well really **nextpnr**) needs to be made aware of the FPGA hardware clock frequency so it can "layout" the FPGA to the required timing budget.

I've also attached a snapshot of the iCEStudio's "Command output" from a design using the Alchitry Cu iCE40-HX8K board (which has a default FPGA clock of 100MHz on the PCB) and you can see a that the **nextpnr** timing budget is still based on a "target Frequency" of 12MHz... (Despite the FPGA actual clock frequency being 100MHz) this leads to many timing issues once the FPGA internal routing starts to get more complex - nextpnr is unaware of the FPGA's real clock frequency (Target frequency)...

[image]

Ideally, iCEStudio should have a selection Tab in one of the dropdown menus to use "Board default clock OR custom FPGA frequency"  (in my case 100MHz) this information is then passed to the nextpnr via the command line when nextpnr invoked by iCEStudio... From what I understand - there is currently no "FPGA Clock frequency" field for each "board" configuration file so iCEstudio cannot pass this information onto the **nextpnr** place and route tool.

As mentioned above, a "workaround" would be a have a user selectable manual configuration option in iCEStudio - the user would enter the FPGA hardware clock frequency - and this is passed by iCEStudio when invoking the nextpnr command line (via the--freq {freq} command line argument).

I understand that adding an new field to the menu tabs requires some GUI work - so a temporary solution for me would be to increase the current default of 12MHz to 100MHz (in the iCEStudio command line that involves nextpnr) :)

I'm using windows8 / latest WIP nightly build...

Again thank you for making the time to investigate this issue for me as I cannot progress further without a solution...,

John

cavearr, 2021-11-23:
Thanks a lot John for your explanation, i don't understand well your question at first!, i'm thinking for a good solution because this is something important, give me some time, i'll try things today and update this thread with results or tests that i need that you try with your boards.

John-Westlake, 2021-11-23:
Hi Cavearr,

Thank you - I'll be happy to test...

Obijuan, 2021-11-23:
Please, could you upload your icestudio design for us to try? I have not tried with your Alchitry CU iCE40-HX8K board, but with many others there is no frecuency set to 12Mhz. Maybe it is a bug specific to that board. We have to check it with your design. It will be very helpful to have it for testing

cavearr, 2021-11-23:
Hi John, as @Obijuan says, there is no default 12Mhz setting for frequency timer.

Nexpnr uses 12Mhz as the base seed for its routing algorithms and what you are seeing is a possibly informational message. Starting from this start frequency nextpnr tries to find the maximum frequency, for the layout, after a few rounds it finishes and tells you what the maximum frequency is for your fpga clock signals in your layout. 

If the timing is adjusted with the parameter you indicate, as I have been seeing in the Yosys forums, what it does is to be able to adjust the maximum working frequency more, since the type of algorithms that are used do not ensure you find the best solution And therefore the more information you give it, the better it is supposed to do. But this is not 100% safe or true either, since in algorithms of this type, sometimes adjusting the restrictions too much does not find good solutions.

What I mean is that in principle this parameter would not be necessary or really important for your problem.

 In the command output, you can look for something like this:

! [image] (https://user-images.githubusercontent.com/6904865/143121559-e26375e7-d6c9-4cba-8e40-68b74b9a69fb.png)

Those would be the maximum clock speeds at which your circuit can operate, so if your design indicates speeds of 100 Mhz or more, it means that your circuit should work perfectly at those speeds.

As @Obijuan says, it would be interesting if you could send us the design or the fragment of the design that causes you problems.

We need to understand your problem well in order to help you.

John-Westlake, 2021-11-23:
Hi Obijuan / Cavearr 

Currently my incomplete design uses about 50% FPGA resources (HX8K) - but the Max Frequency is already around down to 58MHz...

From my understanding, nextpnr will become more aggressive with the routing timing when required to by a higher "Target frequency" constraint - but for sure routing to 12MHz timing target is very lax, and the routing algorithm's will not make much effort at optimization passes...

Talking with those who use Lattice's IceStorm - and they say that the correct target frequency is important to meet higher frequency operation... with Icestorm's FPGA router routines becoming far more aggressive and as a result more optimized (but for sure taking longer to complete the routing process)...

Clifford Wolf in one of his presentation of nextpnr said it performed just as well as Lattice's own toolchain with regards to timing margins, so I would fully expect that configuring the correct Target frequency is also important with nextpnr (as it is with Lattices own design tools).

The simple test would be to set the target frequency to 100MHz in the nextpnr command line with the --Freq{} argument and then compare results with the "default" Frequency argument...

John

cavearr, 2021-11-23:
Hi john, i'm asking you privately for some personal data of your installation, i'm try to patch your apio installation to permit you pass this argument and in this way , you could tell us if it works.

cavearr, 2021-11-24:
Hi again @John-Westlake !  i have some tests and i think is very easy that you could try if it works.

You need to locate the file:

**lib/python3.9/site-packages/apio/resources/ice40/SConstruct**

inside your venv directory into the .icestudio user dir (in my case in linux for example is _/home/carl/.icestudio/venv_)

If you activate the debug mode and review icestudio.log find, you could fond very easy your installation paths.

also, you could try to search in windows finder (search for SConstruct for example).

python3.9 is in my case, in your case depends your python version.

When you find this file, edit it and go to the line indicated in this image and add the freq param in this way:

[image]

Save and restart icestudio, and your nextpnr use it!

[image]

Try to test it an tell us!

John-Westlake, 2021-11-24:
Good morning, sorry for the delay, been fighting a dead car battery that let me down outside the shipping center this morning...! 

I'll compare (and post here) results of 12MHz / 100MHz target freqency...

John-Westlake, 2021-11-24:
OK, results:-

Original run (with 12MHz target frequency):

[image]

12MHz Build "Passes" without error message - but hardware fails... The "Max Freq" at the bottom corner of iCEStudio indicates 250MHz???
 
Build with SConstruct modified for 100MHz:- "Failed error message" & the "Max Freq" at the bottom corner of iCEStudio is blank...

[image]

[image]

[image]

[image]

So the build with SConstruct  set for 100MHz has slightly "Worst case" timing results (17.5nS  verses 16.36ns) for SConstruct @ 12MHz - each build is slightly different.

But at least with SConstruct correctly set for the FPGA boards clock frequency, the 100MHz build "Fails" rather then indicating a Max Freq of 250MHz (lower corner iCEStudio) - which is the case when SConstruct set @ 12MHz.

I'm not sure what the "negative timing budget of x.x ns" in the 100MHz Command Output refers to...

Also to note on my design is that all 8 available Global clock buffers are used, but I have no way of knowing how iCEStudio has assigned the Global Clock buffers (I only have one input clock and no PLL)...

cavearr, 2021-11-24:
Hi John! thanks for your tests. The kind of routing algorithm that nextnpr uses, in most of the cases will do worst results  if you fix the freq. 

This is because this kind of algorithms  are based in the concept of entrophy and if you fix a lot the target, it can't generate enougth randomness to look for the best result.

In this way each execution will be different.

The 250Mhz of your design is for some clk signal in concrete, but you need improve your design to get in all of your clocks the minimun of 100Mhz.

If you see in the directory of your design, you view a directory called ice-build that contains a main.v file that is the verilog generated for your design.

The clock buffers are internal "tools" that nextpnr uses to  route the design and assign the correct time to the signals, for example if nexpnr needa generate a flipflop for its routing in some place, it need to add a "clock buffer" to delay the other signals and  synchronize with the delayed by the flip flop, in his way you lose Mhz in things like his. 

For example imagine that nextpnr use some fpga area and it needs to place a new block but there is no space near the others. In this case nextpnr put the block far and needs route a long "cable" to connect with the others, but in this case it needs to put some clock buffers in the occupied area to delay de clock signal until arrive at the far block.

I don't know if i explain well XD

In resume focus in your design and dont be worried fot the freq parameter i think will be worst if you set it.

cavearr, 2021-11-24:
How this issue is not relative to the current  icestudio functionality, i'm marking it as closed, if is it necessary reopen it.

Thanks!

closed 2021-11-24 by cavearr

nandland, 2024-09-20:
I would also really like the ability to change the clock frequency from the default of 12 MHz. This is the most fundamental constraint (other than pins) when working with FPGAs. Many boards run at > 12 MHz and apio is not going to generate correct bitstreams for those boards. They are going to be subject to timing errors.

cavearr, 2024-09-21:
Hi! i'm working in a new core for this autumn that add this feature, i'm working hard to have very soon. I'm maintaining you inform in this thread.