FPGAwars/icestudio #563: Pick-up problem on some block from the collection manager !

jojo535275, 2021-12-16:
Hola everyone,

One week ago, i added few block in the collection icePLL  (recently created by obijuan). 

see collection here   [icePLL-main.zip](https://github.com/FPGAwars/icestudio/files/7729540/icePLL-main.zip)

All these blocks can be correctly picked-up and dragged to the workspace, by selecting the collection (eg: select/collection/icePLL-main) and using the top left part of the menu toolbar.

But when we used the collection manager, most (but not all) of the block cannot be picked up!

I have the same problem on two different machines (win10 and ubuntu virtual machine)

Can you try to reproduce this bug, Any idea why i have it?

Thanks

Joaquim

mSlider, 2021-12-16:
I can reproduce it in Win10.
In the "Default collection" there are 2 blocks (Prescaler22 nad PrescalerN) that don't let me pick up.

I'm gonna to investigate if these blocks have some different with the others or is a problem with the treeview itself.

Alex (mSlider)

jojo535275, 2021-12-16:
yes the same here
[image]

mSlider, 2021-12-16:
Jo, Can you confirm if the blocks that gives errors are those ones that has no SVG defined as the icon???

jojo535275, 2021-12-16:
Hola Alex

No, it is not only those with missing SVG

In fact i have identified at least 2 ways of "breaking a block" (from the point of view of the collection manager)

here is how to do.

1 - open with icestudio the file probe_tool-clock_div_by1000.ice   in folder    C:\Users\{your username}\.icestudio\collections\icePLL-main\blocks\PLLfreqp
remark:   C:\Users\{your username}\.icestudio\collections\   is where icestudio unzip the collections files when you add a collection.

2 - save the file with a new name  (save as)  xxxx.ice   
3 - restart icestudio and check if you can open that block from the CM (it should be ok)
4- open the project info and delete all the text in the description field  ("clock/1000, only for measuringFpga internall fast clock (eg: PLL) frequencies with a low speed logic analyser" )
5- save the file as zzzz.ice and  restart icestudio and check if you can open thaht zzzz.ice block from the CM (it should be not ok)

Remark : if you redo all this procedure but at step 4 you reset de svg icon you get the same broken result!
remark 2 : if instead of fully deleting the description field, you just modify it, the block remains readable from the CM. and same if you change the SVG icon (without resetting it)

Alex, do not forget to have some rest ! ;-)

Joaquim

jojo535275, 2021-12-16:
Oh my god,

You are so fast :-)  i will your fix  and tell you!
Thanks a lot

Joaquim

mSlider, 2021-12-16:
Jajajaj, the "description" problem was next to the "image" problem, fixed the two problems for the same price!!!

Yeah you are right... I'm gonna go to sleep in a few minutes!

Alex (mSlider)

cavearr, 2021-12-17:
Let's rock!!! youare on fire!!!

Thanks a lot team!

cavearr, 2021-12-17:
Solvet at https://github.com/FPGAwars/icestudio/commit/c90a1330dddf0655d1e3639c9a761b5c84ef71ec

jojo535275, 2021-12-17:
Hola guys,

Sorry Alex, but we where a bit to enthusiastic last night :-)
It seams the problem is still there.

For example all the block in the ICE40 sub folder are still not pickable from the CM

When we open one of the ICE40 pll blocks with notepad++, for the icon field, we see:
  "image": ""

So, it may also be a problem to have, in the list of conditions, the code
  &&  obj.package.image !== ''

So, i quickly tried commenting the 2 modified lines in IceGuiTree.js and it seams, it solves most of the problems.

// obj.package.description !== '' &&

// &&
// obj.package.image !== ''

Remark : 
After this, the only unpick-able block are the ice files which placed directly in the block folder of the collection (instead of being placed in a subfolder of block folder). In the icePLL library, it is this one
[image]

The question is: "Should we allow blocks to be placed there?" 
Because those blocks can not, actually, be seen by the other way of picking blocks (with the top bar menu).
I suppose that  this was made to avoid to have a potentially quickly overloaded and unusable top bar menu!

Any way, i think you understood i found to tedious, to pick-up my blocks from the top bar menu ;-).
But maybe some people prefer that way!?
And if nobody uses it, to make think simpler/less confusing, one day we may just suppress the blocks pick-up on the top toolbar menu!? 
But for that the collection manager plug-in should be rock solid/efficient, with a fast loading time even for loading more than 20 collections !

Have a good day

Obijuan, 2021-12-17:
Currently It is not allowed to place blocks directly on the menu bar. They should always be placed inside a folder. Each folder correcpond to a Menu. So, the probe_tool-clock_div_by10000.ice should be moved to a folder (or removed)

jojo535275, 2021-12-17:
Hola Juan

Yes, yesterday, i also put that same block in the PLLfreqprobe subdirectory.
And i will remove the one which is "outside of a folder"!

But what do you think about the idea of removing the block pick-up from the top bar menu? Do you see a big inconvenient for that? 
Of course, this will be a "quite big" change and we can let this potential removal for latter !

mSlider, 2021-12-17:
Morning Jo,

I'll take a look at the problem with the fields containing '' or ""

Thanks for teststing it, we will try to make it rock solid.

Alex (mslider)

Obijuan, 2021-12-17:
> But what do you think about the idea of removing the block pick-up from the top bar menu? Do you see a big inconvenient for that?
> Of course, this will be a "quite big" change and we can let this potential removal for latter !
> 

No big problem. It could be removed. No blocks on the top bar, just menus

Maybe in the future we could add toolbar or something, but for now, in my opinion only menus is ok

mSlider, 2021-12-17:
> No big problem. It could be removed. No blocks on the top bar, just menus
>
>
>
> Maybe in the future we could add toolbar or something, but for now, in my opinion only menus is ok

May be if we remove that block tool bar, we can put there the shortcut buttons... do you see as a good place for that?

jojo535275, 2021-12-17:
Ok , let us open a new issue to discuss about that removal!
https://github.com/FPGAwars/icestudio/issues/566

cavearr, 2021-12-17:
Hi guys! you are incredible!! wohooo!!! i'm reading all your messages now, today i'm little offline!

jojo535275, 2021-12-18:
Hola Guys,

I am trying to load correctly one more collection, last old and unsupported "FPGA-peripherals" collection.
It is actually seen as "invalid" by icestudio !
i made an "injerto" of all the block of thaht collection in good collection and renamed it "fpga-periph". (see joined file)
i now can see it on my icestudio !

But most of the Blocks (have all the fields empty in "project information") are also not pickable from the CM.
eg:   fpga-periph\blocks\DHTxx\DHT22_Serial.ice
or fpga-periph\blocks\i2c\ADC_read.ice
...

To solve this, i commented in IceGuitree.js

      // &&
      // obj.package.description !== null &&
      // obj.package.description !== "" &&
      // obj.package.name !== null &&
      // obj.package.name.length > 0 &&
      // obj.package.image !== null &&
      // obj.package.name !== ""

So the question is 
Shall we make this change  (it allow block creators to let the fields empty). ?
Or should we force the creator to supply  some fields (in that case, it will be preferable also to warn him ,when he saves his .ice file, that he must fill in that "empty"field).

For more user freedom i personally prefer to make the change !

In fact, it will be nice to discuss all together about all the code/condition that impeach the CM to load correctly some blocks.
I mean what is the pertinence of every line between line 273  and line 286 of IceGuitree.js
[image]

[fpga-periph.zip](https://github.com/FPGAwars/icestudio/files/7739431/fpga-periph.zip)

Thanks
Joaquim

mSlider, 2021-12-18:
Hi Jo,
I think FPGA-peripherals is not an ice collection (isn't it?), beacause there are the "serial" and "servos" blocks that are written in verilog .v rather than icestudio .ice.
Likewise I'll try the blocks for see how are they affecting.

I'll try to make an explanation later with some examples for these test cases/filter. To see if something is duplicate, weird or effective relative to the code blocks params.

jojo535275, 2021-12-18:
Yes Alex, i think you are right!

So in the end , Maybe it is better to just move the interesting blocks in an existing collection like the Jedi-collection under varios-3.  To make them accessible from icestudio !

What do you think Juan ?

And we will organise all this with time! A task already started by Juan with the ICExxx collections.

Thanks Alex and a good evening for everybody;
Joaquim

mSlider, 2021-12-18:
Ok, I think the important question here is:
- Do us allow the user can be the "name" and/or "description" fields empty or not???

Since we can answer the question it's easy to establish the filter...

Good Afternot to all.

Alex (mSlider)

cavearr, 2021-12-18:
Hi guys! this is an important thing.

I think the name should be a required field.

Should be very interesting create a new plugin to create collections from icestudio, checking if the blocks are consistent.

If the block has "no name" this has no sense to display in the list of the collection manager i think, what do you think?

the image and descriptioni is not neccesary . I want to do in other versions of colllection manager that you can view the description of the block in some way (in other info panel or something similar), but for the moment is not neccesary.

If you want , for me only make te¡he name a required field, and if you think we can make a dialog that alert of the blocks without name, or putting in the tree with an alert icon for example (what do you think?)

I'm working in the database persistence this weekend and i could do this fixes.

Tell me what do you think!

jojo535275, 2021-12-19:
Ola Carlos,

Correct me if am wrong, this "Name" set in "the project information" window is **only used** to be shown on top of the block in the workspace (when no SVG was also set).
And no "name" (and no SVG) give us that "ugly" ,"guess who i am" :-) result!
[image]

So, i agree with you that we need at least the "name" field filled.

But, why not using the same name as the .ice file to fill automatically that field (just removing the ice extension)
In fact this is what the collection manager do to show the block in his tree-view!

So in the end we will have only ** a single name for each block** the one chosen by the "saves as" function!

Eg:   the block **ADC_read.ice** will appear as
   - **ADC_read** in the CM tree-view ,
   - **ADC_read**  in the "project information" name field of that block 
   - and **ADC_read**  shown in the bloc graphics on the wok spaces ( if no SVG set).

And this also means, that when we use the menu "File/add as a block" function, the imported block will automatically be renamed (in project information of that newly created block) with the name of the source .ice file. Actually it is loaded without name and we have to fill in manually the project information name field.

I think this "single name simplification" will remain compatible with the block "clone" function!
Be as critic as possible about that simplification idea, Because, as a quite new Icestudio user, i may just miss something that make-it irrelevant by breaking another icestudio function!

And if there is a reason for not doing that "single name simplification", 
i agree with your ideas, Carlos, for forcing/warning the user to fill the Name field !

Remark: All this thinking gave me an idea for renaming/"reworking-slightly" the clone function (making it more intuitive) 
i may open a new issue on github to discuss about that in the next days!

@ Alex: i will  go to sleep few hours, then i will try you code suggestions!

Have a good day,

mSlider, 2021-12-19:
Hi,
In my opinion "filename.ice" must be the same as the "field name".

Problem is if you rename the file externally, the "field name" is not updated.

Alex (mSlider)

cavearr, 2021-12-19:
The problem with the filename is that @mslider said. This is easy to fix, but i think we dont use this because the idea is that blocks could be translate and for translate the filename only could have one :) 

I think the best solution is force to add a name to the block once is created. 

for example you create a new block and when you back to the main design, if name is not present icestudio force a dialog to fill (or open the block info popup to take the required fields. 

I think this is a better solution for consistent blocks. 

Enviados: Domingo, 19 de Diciembre 2021 10:05:16

mSlider, 2021-12-19:
Hi @cavearr,

I don't uderstand this you said:

> This is easy to fix, but i think we dont use this because the idea is that blocks could be translate and for translate the filename only could have one :)

Can you explain it better?

jojo535275, 2021-12-19:
Hola Guys,

@ Carlos, correct me if i am wrong, but actually the "translation" function is not really fully implemented for blocks!
for example the names shown in the CM tree-view are the .ice filename (minus the .ice extension)

For translations, If it will be the same logic as for icetudio GUI translation, we will have one .po file for each language in a collection folder (eg: in /iceCounter/locale/  we will have (en_EN.po, es_ES.po,...) ).

@ Carlos wrote "Problem is if you rename the file externally, the "field name" is not updated."
You are right Carlos, but if instead of storing all the translations in po files (en_EN, es_ES,...).
we **store all the translations concerning a block inside the .ice file** the problem will be solved.

An advantage for that is also that deleting an .ice file will not require to open all the po files for removing the translation concerning that block.

But Personally: i think that block translation is really to much work for collection makers ! And it doesn't worst the effort! 
Icestudio Gui translation is sufficient for me ;-)

Have a good evening

Joaquim

mSlider, 2021-12-19:
Hi @jojo535275 

What things in a block can be translated?
I ask this because I think the name is a parameter that must be have not translation because is a "Name".

Other concept problems I don't see clear:
1. What is the diference between project and block? Because the only way you can fill the "name" field is in "project Information" tab.
2. The other concept problem I see is the "name" field must be "title" because name must be unique and is implicit in name.ice

Hope you understand my confusion and can be solved...

Alex (mSlider)

cavearr, 2021-12-19:
Hi @mslider, project and block is basically the same, the only difference is that in a block you can't have IO pins configured (because should be generic for all of boards).

About the "easy fix",  i'm referer to  use the filename as name block, i try to explain that this change is very easy but i think this is a solution for a "short term problem" but is not scalable for future enhancements.

@jojo535275  i don't recomend that use the filename as the block name because this is a very inconsistent point, i think all should be managed inside icestudio.

About the translations are important, although now there are a lot of blocks without translations, in near future i'm sure should be very neccessary and we need to prepare all for this moment.

Your idea that translations will be inside the .ice is very interesting and i'm very agree with it, maybe we need to redesign the info block popup, i'm thinking on it since this discussion. I have started working in a new concept of  configuration panes, like kikad config tables, that i think are very usefulls, i'll try to make a mockup to review with you.

I'm thinking in this and the toolbar, i'll try to have proposals soon.

jojo535275, 2021-12-19:
Hola Alex, and Carlos,

If we do the "block translation" (mentioned Carlos).
We can of course try translating.
- the "description" field
- And  what Carlos want, i thinks is also translate the name which appears on the block graphics;
[image]

And this name is now set with the "Name" field in the "project Information" window. 

About translation, maybe it is good **first for you to understand how the icestudio Gui works!**

an example: at in the icestudio/views/ menu.html you can see at line 48 the code for translating the "File" text on the top bar menu.
[image]

and on github repository icestudio/app/resources/locale/es_ES/ you can find (eg: the spanish) .po translation file which contains at line 499 (the default code language) & at line 500 (the users selected language):
[image]

As i wrote in my last message for the block translation i supposed it will work similarly! 
Maybe Juan or Carlos can give us some more info about the difference between blocks translation and Gui translation?

About your point 1 : the difference between a project and a block, it is an important question, Carlos just gave you some explanation about it and i will add that:

- a project is a "top level" design (it has a project information window)
- and a block was a top level design (that was saved in an .ice file) and that we now import in another top level design. 
So this loaded block becomes now a sub level design! 
And for me, it will be better to have for those block a "block information" windows instead of a "project information" window
 and i have some ideas of information that we can show differently in those "Block information window" to make thinks less confusing ! (eg: a "modified" mark when a block is different from the on present in the original/ collection source ice file.)
Maybe these joins a bit what Carlos wrote one hour ago about modifying the pop-up windows.

About your point 2, yes renaming the name field as "title" could be a slightly better solution, but maybe not the best one!

i understand your confusion Alex, as a 10 month icestudio user, i can use it with no problem, but still fell a bit uncomfortable by some things concerning the blocks stored in a collection, the imported and, modified blocks and the cloned blocks !

This github issue is starting to be long and it is sliding a bit to other important subjects/issues. 
Where is the best place to discuss about "wide" subject like this one?

Will it be better to discuss about the subject concerning the filename, "Name" field , "project information" windows in a fpgawars google group new thread or a github new issue ? 
or shall we continue here? 

Thanks and have a good night guys!

Joaquim

cavearr, 2021-12-19:
Hi Jo! maybe i think that it's  a great idea translate this discussion to FPGAgroup and with conclussions, make a new post in this thread.

I think is a great idea because more preople could discuss andgive their opinions.

Could you launch the message to FPGAgrous ,starting the discussion, and referencing this thread for all of they that  would like to read?

Your feedback as users is very important, the idea is convert icestudio in a very nice , userfriendly tool and a lot of concepts are very confussed now. 

About the translations, the gui translations could be edit by poedit tool, are you try it? it's very simple and powerful.

The gui translations for me are in good way with the i18n support and the po files.

The collections transaltions works in  the same way, i dont explore it, @Obijuan ,could you illuminate this?  but i think for blocks probably is more easy if translations are in the same place that block info.

In the near future, you can select or attach to block info , examples for this block and in the collection managar you could see the description for the block and a list of its examples (for example splitting in vertical the collection manager win a 2/3-1/3 proportions for example and in the lower aream put the selected block descriptions and the links to its examples (or if you don like this display, we can do a right tab action with a menu over the block with this kind of options...)

I dont know if you like it buar are some ideas. Anyway i think is important to open a new thread in FPGAwars to discuss about all of this things, with @Democrito and the others.

Thanks team!

jojo535275, 2021-12-20:
Ok Carlos, i will open a thread in the FpgaGroup!

Yes, i know that we have all the common objective of making icestudio even easier to use ! 
It is just sometimes difficult to explain what we have in our mind and to understand what is in other peoples mind :-)

But we will manage!
Yes i already used poedit for the Gui translation few month ago and this was ok! 
For block translation, if storing them in the.ice file, we will maybe need another tool than poedit. maybe an icestudio plugin that can write in the ice file.
About the near future functions, we already discussed in the past about some of them and yes they look interesting !

Thank Carlos an Alex

So let us go to the FPGAward the group [new thread](https://groups.google.com/g/fpga-wars-explorando-el-lado-libre/c/LLw7WkKimTM)

Obijuan, 2021-12-20:
The process of translating a collection is docume ted here:
https://github.com/FPGAwars/iceK/wiki/Translating-the-collection-into-your-language

(It is shown for the icek collection but it is the same for the others)

The translations only take effect if the collections are installed (not valid for external collections)

You can try it: install any stable collection (icek or icewires). The default language is English. If you change to Spanish all the texts are translated (information, descriptions...)

I usually do not translate the blocks names, but it could be done

It would be great if you try to translate to other languages. For example, @jojo535275 could try to translate some texts of the icek collection into French, for testing)

Currently I am working on the iceIO collection. There will be a stable version on christmas holydays

I think that currently it does not worth it to insert the translations into the .ice blocks. The current .po system could not be the best one but it just simply works. I prefer to spend the time on other issues

I remind you that I am in the process of reorganizing all the collections. The collections that have already been reorganized are called stables. What I am doing is:

* Translating all the blocks into English. This is the default language
* Implementing the blocks using only other blocks (from stable collections). Only the foundamental blocks are in verilog (flip-flops, gates, io configurations...)
* Storing the svg icons into the collection project
* Documenting the collections (in English) on the wik
* Spliting big collections into smaller ones (like the Jedi collection, which is huge)
* Releasing the stable collections

Current stable collections are:
* icek: For constants
* icewires: for wires/buses

The Next stable collection will be the iceIO (and the next one the iceGates)

jojo535275, 2021-12-20:
Hola Juan,

Thanks a lot for this interesting and detailed description about the actual situation of the collections translations! 
it was just what we need to know.

I will try a bit all of this and reply/make some question in the FPGAwars group!
Muchas gracias and have a nice day.

Obijuan, 2022-01-17:
I think this issue can be closed now

closed 2022-01-17 by Obijuan