FPGAwars/icestudio #589: Error executing command set

TheSnowfield, 2022-01-12:
[image]

I have this problem while I installing the stable toolchain. My board is `iCESugar-nano`.
How to deal with this problem? or any logs? I cannot copy this message, and the path on the message is incomplete.

Any idea. Thanks!

jojo535275, 2022-01-12:
Hello TheSnowfied,

Maybe try installing the last wip  0.8.2w202201110401    at  https://downloads.icestudio.io/

To copy the message, you can open the  View / command output window from the top bar menu!

TheSnowfield, 2022-01-12:
> Hello TheSnowfied,
> 
> Maybe try installing the last wip 0.8.2w202201110401 at https://downloads.icestudio.io/
> 
> To copy the message, you can open the View / command output window from the top bar menu!

Sadly, seems it doesn't work for me.
icestudio hangs on installing `oss-cad-suite` package. 

```
>>(OUTPUT): Successfully installed apio-0.8.0 attrs-21.4.0 blackiceprog-2.0.0 certifi-2021.10.8 charset-normalizer-2.0.10 click-8.0.3 colorama-0.4.4 icefunprog-2.0.3 idna-3.3 intelhex-2.3.0 jsonmerge-1.8.0 jsonschema-4.3.3 packaging-21.3 pyparsing-3.0.6 pyrsistent-0.18.0 pyserial-3.5 pyusb-1.2.1 requests-2.27.1 scons-4.2.0 semantic-version-2.8.5 six-1.16.0 tinyfpgab-1.1.0 tinyprog-1.0.21 tqdm-4.62.3 urllib3-1.26.8 wheel-0.37.1

>>(ERROR): WARNING: You are using pip version 21.2.4; however, version 21.3.1 is available.
You should consider upgrading via the 'C:\Users\CirnoBaka\.icestudio\venv\Scripts\python.exe -m pip install --upgrade pip' command.

**** STEP: APIO install oss-cad-suite
>>>> utils.executeCommand => set APIO_HOME_DIR="C:\Users\CirnoBaka\.icestudio\apio"& set PATH=C:\Users\CirnoBaka\.icestudio\venv\Scripts;F:\Python39\Scripts\;F:\Python39\;F:\VulkanSDK\1.2.176.1\Bin;C:\Program Files\Java\jdk1.8.0_191\bin;C:\Program Files\Java\jdk1.8.0_191\jre\bin;C:\Program Files (x86)\VMware\VMware Workstation\bin\;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.2\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.2\libnvvp;F:\MentorGraphics\9.5PADS\SDD_HOME\common\win32\bin;F:\MentorGraphics\9.5PADS\SDD_HOME\common\win32\lib;F:\MentorGraphics\9.5PADS\MGC_HOME.ixn\bin;F:\MentorGraphics\9.5PADS\MGC_HOME.ixn\lib;C:\Program Files (x86)\SCE\PSP2\Tools\Publishing Tools\bin;C:\Program Files (x86)\Common Files\Intel\Shared Libraries\redist\intel64\compiler;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;F:\NetSarang\Xshell 6\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Program Files\Git\cmd;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\Nintendo\Nintendo Dev Interface;C:\Program Files\NVIDIA Corporation\Nsight Compute 2020.3.0\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\nodejs\;F:\Python37\Scripts\;F:\Python37\;C:\Users\CirnoBaka\AppData\Local\Microsoft\WindowsApps;C:\Users\CirnoBaka\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\CirnoBaka\.dotnet\tools;F:\MiKTeX 2.9\miktex\bin\x64\;F:\MentorGraphics\9.5PADS\SDD_HOME\CAMCAD;C:\Users\CirnoBaka\AppData\Roaming\npm;F:\iverilog\bin;F:\iverilog\gtkwave\bin;F:\NVIDIA Corporation\Cg\bin;F:\NVIDIA Corporation\Cg\bin.x64;F:\CMake\bin;C:\Users\CirnoBaka\AppData\Local\Programs\Python\Python38\Scripts\;C:\Users\CirnoBaka\AppData\Local\Programs\Python\Python38\;F:\Python37\Scripts\;F:\Python37\;C:\Program Files (x86)\VMware\VMware Workstation\bin\;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.2\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.2\libnvvp;;F:\MentorGraphics\9.5PADS\SDD_HOME\common\win32\bin;F:\MentorGraphics\9.5PADS\SDD_HOME\common\win32\lib;F:\MentorGraphics\9.5PADS\MGC_HOME.ixn\bin;F:\MentorGraphics\9.5PADS\MGC_HOME.ixn\lib;C:\Program Files (x86)\SCE\PSP2\Tools\Publishing Tools\bin;C:\Program Files (x86)\Common Files\Intel\Shared Libraries\redist\intel64\compiler;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;F:\NetSarang\Xshell 6\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Program Files\Git\cmd;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\Nintendo\Nintendo Dev Interface;C:\Program Files\NVIDIA Corporation\Nsight Compute 2020.3.0\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files (x86)\Windows Ki;C:\Users\CirnoBaka\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\CirnoBaka\.dotnet\tools;F:\iverilog\bin;F:\iverilog\gtkwave\bin& "C:\Users\CirnoBaka\.icestudio\venv\Scripts\apio.exe" install oss-cad-suite@0.0.7

>>(OUTPUT): (DEBUG) Profile path: C:\Users\CirnoBaka\.icestudio\apio\profile.json
(DEBUG) Home_dir: C:\Users\CirnoBaka\.icestudio\apio
Error: [Errno Expecting value] Not Found: 0

----!!!! ERROR !!!! -----
CMD: set APIO_HOME_DIR="C:\Users\CirnoBaka\.icestudio\apio"& set PATH=C:\Users\CirnoBaka\.icestudio\venv\Scripts;F:\Python39\Scripts\;F:\Python39\;F:\VulkanSDK\1.2.176.1\Bin;C:\Program Files\Java\jdk1.8.0_191\bin;C:\Program Files\Java\jdk1.8.0_191\jre\bin;C:\Program Files (x86)\VMware\VMware Workstation\bin\;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.2\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.2\libnvvp;F:\MentorGraphics\9.5PADS\SDD_HOME\common\win32\bin;F:\MentorGraphics\9.5PADS\SDD_HOME\common\win32\lib;F:\MentorGraphics\9.5PADS\MGC_HOME.ixn\bin;F:\MentorGraphics\9.5PADS\MGC_HOME.ixn\lib;C:\Program Files (x86)\SCE\PSP2\Tools\Publishing Tools\bin;C:\Program Files (x86)\Common Files\Intel\Shared Libraries\redist\intel64\compiler;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;F:\NetSarang\Xshell 6\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Program Files\Git\cmd;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\Nintendo\Ni [... cut ...] :\Program Files\Git\cmd;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\Nintendo\Nintendo Dev Interface;C:\Program Files\NVIDIA Corporation\Nsight Compute 2020.3.0\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files (x86)\Windows Ki;C:\Users\CirnoBaka\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\CirnoBaka\.dotnet\tools;F:\iverilog\bin;F:\iverilog\gtkwave\bin& "C:\Users\CirnoBaka\.icestudio\venv\Scripts\apio.exe",install,oss-cad-suite@0.0.7
```

TheSnowfield, 2022-01-13:
Fixed. This problem is from my network environment, apio cant install `oss-cad-suite` successfully.
After some works, I can get apio to work properly and slow downloading.
I guess apio should allow users to install their package from the local disk directly or allow users to use their proxies is better.
[image]

TheSnowfield, 2022-01-13:
@jojo535275 thanks a lot! :3

closed 2022-01-13 by TheSnowfield

jojo535275, 2022-01-13:
Well, my help was not very efficient! :-)
and now i remember thaht there is also procedure for[ installing apio outside icestudio](https://github.com/FPGAwars/icestudio/wiki/Installing-the-Toolchains-from-outside-icestudio)  thaht could have helped you

Never mind you got it working!  Congratulations

Have a nice day or evening in Asia ;-)