FPGAwars/icestudio #590: WinError 2

Avocodude, 2022-01-12:
Hi! Im using Ice studio for college classes and am unable to use the program. Every time I try to verify I get this error message.
[image]
Any help would be great!
Thank you.

TheSnowfield, 2022-01-12:
Please check icestudio log file located at `C:\Users\*\icestudio.log`.
This issue seems the same with #589, it causes icestudio freeze and hangs on installing `oss-cad-suite` package.

Avocodude, 2022-01-12:
Hi! TheSnowfield, So once I find icestudio.log, what should I do next?

TheSnowfield, 2022-01-12:
> Hi! TheSnowfield, So once I find icestudio.log, what should I do next?
Scroll to end of file and find some error logs, paste to here. 🤣

Avocodude, 2022-01-12:
Error flag: false
  No Errors. Toolchain installed--
tools._executeLocal>set APIO_HOME_DIR="C:\Users\alexr\.icestudio\apio"& "C:\Users\alexr\.icestudio\venv\Scripts\apio.exe" verify --board icestick -p "C:\Users\alexr\OneDrive\Desktop\DGFN II\ice-build\DGFN Lab 0"

jojo535275, 2022-01-13:
Helo Avocodude and TheSnowfield,

Last week we discussed about[ this error](https://groups.google.com/g/fpga-wars-explorando-el-lado-libre/c/t22QPTZefS0/m/kXnV6e1fCQAJ)
Apparently it is related to apio toolchain.

And Obijuan suggested to do follow these steps:

1. Open icestudio (the same version you were using previously)
2. Remove the toolchains (Tools/Toolchain/Remove)
3. Install the toolchains (it will install the updated apio packages). The apio version is the same than before: 0.8.0 (but the apio packages have been updated)
4. Open any circuit and try Verify

What version of icestudio  (wip) did you instalalled ?

Regards

Avocodude, 2022-01-13:
It works! Thank you guys so much for your help. Yall really saved me!
Thanks again!!

closed 2022-01-13 by Avocodude