FPGAwars/icestudio #610: startup fail (Linux mint 20.2)

ermtl, 2022-01-15:
I tried to install version 0.8.0 and  the latest nightly on a Linux Mint 20.2 machine and got the following error.
[Image](https://dox.mailpeers.net/IceStudio1.png)
Steps to reproduce :
- Remove the old .icestudio directory : rm -rf ~/.icestudio
- download the latest appimage (0.8.2w202201110401)
- Make it executable
- start it
- select the board
- Get that error
[Image](https://dox.mailpeers.net/IceStudio1.png)

Unsurprisingly, if I then try to install the toolchain I get this new error:
[Image](https://dox.mailpeers.net/IceStudio2.png)

I have Python 3.8.10

cavearr, 2022-01-15:
Hi! could you send us the icestudio.log that was created in you home directory? you could see detailed instructions in issue #494

ermtl, 2022-01-15:
I have done it. at first, the log mentioned missing dependency venv, I installed it using : apt install python3.8-venv

Still the same error, .icestudio only contains a file named profile.json, there is no venv/bin directory, creating it manually does not fix the error. Here is the log:

`
=======================================================================================
 Icestudio session Sat Jan 15 2022 16:57:46 GMT-0500 (Eastern Standard Time)
 Version: 0.8.2w202201110401
=======================================================================================
Node information:
  * Node version: v16.4.0
  * lts: undefined
  * SourceURL: https://nodejs.org/download/release/v16.4.0/node-v16.4.0.tar.gz

System information:
  * Architecture: x64
  * Platform: linux

Profile file: /home/eric/.icestudio/profile.json

- PROFILE:

{
  "data": {
    "board": "icestick",
    "boardRules": true,
    "collection": "/tmp/.mount_7EFY6V/usr/bin/resources/collection",
    "language": "en",
    "uiTheme": "dark",
    "externalCollections": "",
    "externalPlugins": "",
    "remoteHostname": "",
    "showFPGAResources": false,
    "displayVersionInfoWindow": "yes",
    "lastVersionReview": "0.8.2w202201110401",
    "loggingEnabled": false,
    "loggingFile": "",
    "pythonEnv": {
      "python": "",
      "pip": ""
    }
  }
}

- PATHs

common.BASE_DIR: Icestudio base dir: /home/eric
common.ICESTUDIO_DIR: Icestudio folder: /home/eric/.icestudio
common.PROFILE_PATH: Profile path: /home/eric/.icestudio/profile.json
common.INTERNAL_COLLECTIONS_DIR: Internal collections: /home/eric/.icestudio/collections
common.APIO_HOME_DIR: APIO folder: /home/eric/.icestudio/apio
common.ENV_DIR: PYthon virtual environment: /home/eric/.icestudio/venv
common.ENV_BIN_DIR: Executable files: /home/eric/.icestudio/venv/bin
common.ENV_PIP: PIP executable:  /home/eric/.icestudio/venv/bin/pip3
common.APIO_CMD: Apio command: APIO_HOME_DIR="/home/eric/.icestudio/apio" PATH="/home/eric/.icestudio/venv/bin:/tmp/.mount_EMtjC1/usr/bin:/tmp/.mount_EMtjC1/usr/sbin:/home/eric/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" "/home/eric/.icestudio/venv/bin/apio"
Common.APP: Icestudio APP folder: /tmp/.mount_EMtjC1/usr/bin
common.APP_DIR: Icestudio execution folder: /tmp/.mount_EMtjC1/usr/bin

===> tools.CHECKTOOLCHAIN
>>>> utils.executeCommand => APIO_HOME_DIR="/home/eric/.icestudio/apio" PATH="/home/eric/.icestudio/venv/bin:/tmp/.mount_EMtjC1/usr/bin:/tmp/.mount_EMtjC1/usr/sbin:/home/eric/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" "/home/eric/.icestudio/venv/bin/apio" --version

>>(ERROR): /bin/sh: 1: /home/eric/.icestudio/venv/bin/apio: not found

----!!!! ERROR !!!! -----
CMD: APIO_HOME_DIR="/home/eric/.icestudio/apio" PATH="/home/eric/.icestudio/venv/bin:/tmp/.mount_EMtjC1/usr/bin:/tmp/.mount_EMtjC1/usr/sbin:/home/eric/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" "/home/eric/.icestudio/venv/bin/apio",--version
  Error flag: true
  Error: true`

cavearr, 2022-01-15:
Could you open a terminal and execute the command:

python -V

and paste the output here?

ermtl, 2022-01-16:
At first I got :
`
~$ python -V

Command 'python' not found, did you mean:

  command 'python3' from deb python3
  command 'python' from deb python-is-python3

`
So I installed with: apt install python-is-python3

The output of the python command is now : Python 3.8.10

Still the same error, log unchanged.

Obijuan, 2022-01-16:
Hi ermtl,

The first error (image1) is because the toolchain is not yet installed (This error happens always the first time you execute icestudio. It is not a real error and will be revemoved soon). Let's focus on the sencond error

I need you to follow this steps for getting more information about the problem:

1. Close icestudio (if it was opened)
2. Remove the ~/icestudio.log file
3. Open Icestudio
4. You will see the initial error. Ignore it
5. Install the toolchain eigher by clicking on the green notification or on the "Tool/Toolchains/Install" menu option
6. It will start the downloading and installation of the toolchains. Wait until the error shows up
7. Upload the new icestudio.log (Provide the full file. Just drag and drop here). I will download this log file and tell you what actions to do

Thanks for your report

ermtl, 2022-01-16:
I still had the second error message, but this time the toolchain install worked (but with some errors still in the log like 'ERROR: Failed building wheel for blackiceprog').
The main differences are that I installed python3.8-venv and python-is-python3

It would be really good to have more explicit, action oriented error messages, also the menu item says 'logging enabled' but you still have to click on it to activate it. People see it's enabled, but it does not work. would be better if the text was 'logging disabled' when it's disabled.

Here is the log, thanks for your help !

[icestudio.log](https://github.com/FPGAwars/icestudio/files/7878101/icestudio.log)

Obijuan, 2022-01-17:
Thanks for the log. Everything is fine and the toolchains have been successfully installed. So, now you should be able to load circuits in your board

Let's recap:

* **Error 1**:  It was not the problem. But it has been corrected on the latest development vesion (0.8.2w202201160601)
* **Error 2**:  The command python3.8 -m venv could not be execute because the package python3.8-venv was missing. Once it is instaled, the error is gone

(The package python-is-python3 is not really needed. If you do not have it installed it will also run fine)

I will close this issue as all the errors reported has been fixed. Thanks for your help

closed 2022-01-17 by Obijuan

ermtl, 2022-01-20:
@Obijuan : I tried on a different machine (Linux Mint 19.3) and it also failed while installing the toolchain but for a different reason. 
Looking at the log, it mentioned pip being the wrong version, so i updated it :

~$ /home/eric/.icestudio/venv/bin/python3.9 -m pip install --upgrade pip
Collecting pip
  Using cached pip-21.3.1-py3-none-any.whl (1.7 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 20.2.3
    Uninstalling pip-20.2.3:
      Successfully uninstalled pip-20.2.3
Successfully installed pip-21.3.1

Trying to install the toolchain failed again producing the same message.

Here is the [picture](http://dox.mailpeers.net/IceStudio3.png)

You'll find the log below
[icestudio.log](https://github.com/FPGAwars/icestudio/files/7902754/icestudio.log)

Also (but I think it's unrelated), during startup and before the splash screen appears,
the following dialog appears for a split second: [picture](http://dox.mailpeers.net/IceStudio4.png)
preference seem to be saved normally.