FPGAwars/icestudio #611: File extension

zipotron, 2022-01-17:
Hello, I just detected that the file extension and the "code tagging" in Github of this project is interfering with another.
Check this issue: https://github.com/sylefeb/Silice/issues/206#issue-1105465885
Also I could see that Github is identifying the IceStudio code as a "Slice".
Thanks for this awesome project.

cavearr, 2022-01-17:
Hi @zipotron!! thanks for advice! Silice is an awesome project and we collide with the same extension.

I'll try to investigate how to fix with Github.

I'm maintaining this thread open while i'm investigating.

Thanks again.

zipotron, 2022-01-17:
With pleasure! The people of Silice seams to move their file extension (as can be seen in their issue) Good luck with the Github parser!

cavearr, 2022-01-17:
yes! i'm reading the thread, for the moment Icestudio maintains the .ice extension, maybe in the future could change for other that not collides.

closed 2022-01-17 by cavearr

zipotron, 2022-01-18:
No needed to change anymore...