FPGAwars/icestudio #612: upload issue with

Pernaci, 2022-01-17:
Hi all,

when I try the function "upload" with the FPGA ORANGECRAB-r02-25f develop board I've the following issue:

"There are errors in the design...... Error 'NoneType' object has no attribute get"

Please can you help me to find a solution?

Thank you

Obijuan, 2022-01-17:
Could you please provide a screenshot of the circuit you are trying to upload? I need to check the details. Thanks

Obijuan, 2022-01-17:
Bug confirmed. I've been able to reproduce it with the Basic One LED example
* The verification is ok
* The Building is also ok
* The upload command generates the error

The bug can be reproduced without having an orangecrab board connected

Obijuan, 2022-01-17:
The bug is in apio 0.8.0. It has been solved in the development version (apio 0.8.1)   
I will keep this issue open until apio 0.8.1 is stable and integrated into Icestudio   
Thanks for you feedback

Pernaci, 2022-01-18:
Hello,

I' ve tried to upload with (apio 0.8.1).
But there is an other issue (Error 74) that you can see in the attached image.
What Can I do? 

Thank you
Chris
[image]

Pernaci, 2022-01-18:
I add that the issue is available also without the board connected. 

thank you
Chris

cavearr, 2024-02-05:
this issue is very old and we update the toolchain to the last oss cad suite in the last wips, you could download from github actions or more easily from https://downloads.icestudio.io

I'm closing the issue but reopen it if you consider it

closed 2024-02-05 by cavearr