FPGAwars/icestudio #613: Problema al caragar una programa en Alhambra II con mi mac

opelectron, 2022-01-28:
Seguro que es un problema recurrente pero tengo problemas para poner operativa mi Alhambra II en mi mac. Cuando intento cargar un programa con icestudio .

He seguido todos los hilos pero no consigo solucion .

Saludos.

[image]
me sale el siguiente error

opelectron, 2022-01-28:
He actualizado python 3.9
He actualizado pip3 i apio 0.8.1

Icestudio no detectecta la placa , no aparece en el listado de puerto .
versión estable
version no estable

Ya no se que hacer.

macos bigsur 11.6.2

cavearr, 2023-06-29:
Cierro este issue, si persiste el error o necesitas ayuda , reábrela,

closed 2023-06-29 by cavearr