FPGAwars/icestudio #618: Build issues and lib8

AsynchronousParallel, 2022-02-18:
When I try to build my very simple turn on an LED project an error message appears.

libintl.8.dylib. not found 

I am running on a Mac and using an ICESugar nano

BitBangingBytes, 2022-02-19:
I have the same issue, Mac M1 with a ULX3S board.

cavearr, 2022-02-20:
Hi!, what version of iceStudio are you using?

AsynchronousParallel, 2022-02-20:
v 0.9.1 w 202202101002

cavearr, 2022-02-20:
Hi! i think this error is for osx security system, you could allow only to Icestudio (this is embarrasing process because osx ask for all the libs, included libintl.8.dylib

Other option is that you disable the security system.

This guide could be usefull:

https://support.apple.com/en-us/HT202491

if you allow iceStudio and  continues not working tell us and we tray to identify the bug.

Thanks!

BitBangingBytes, 2022-02-20:
I can confirm it's only this library and it's not related to OSX security. I have tried the icestudio fork for Mac M1 as well and this is the only library the system can't find. This lib is in another location and if I try to symlink it I get an error saying it's for the arm arch and I need x86 which isn't true since I am on the ARM Mac.

cavearr, 2022-02-20:
Thanks for the information! i'm working in an ARM mac too and all works well, i'll trying to reproduce your problem. 

If i need some test, i'm telling you. 

Enviados: Domingo, 20 de Febrero 2022 23:51:56

BitBangingBytes, 2022-02-20:
Thank you! I have done a clean install of homebrew several times and still it doesn't work. Perhaps you have some other program installed that is supplying that library?

cavearr, 2022-02-20:
yes ! i'm looking in this direction.

Try with this two commands:

brew install gettext
brew link gettext --force

BitBangingBytes, 2022-02-20:
No, still can't find the library anywhere it checks. I tried that, saw it somewhere else with a google search. What location is your libintl.8.dylib file?

BitBangingBytes, 2022-02-20:
So I just got it to Build! GTKWave has the lib and I symlinked it.

sudo ln -s /Applications/gtkwave.app/Contents/Resources/lib/libintl.8.dylib /usr/local/lib/libintl.8.dylib

When I go to upload though now it hangs, I see the blue LED turn on showing it's downloading on the ULX3S but it never completes. If I unplug the board from the USB port then it says complete.

cavearr, 2022-02-21:
Thanks for the tests! i'll try to include this lib in future releases, this lib is not included in Icestudio, because this is used by the toolchain managed by Apio, but i'll try to package it.

About the usb issue i'll try to investigate it.

cavearr, 2022-02-22:
Hi! we add libintl.8.dylib to the toolchain, could you remove icestudio and reinstall it to see if it works from the beginning?

Thanks for your tests and feedback 

(i'm continuing working in your usb issue)

BitBangingBytes, 2022-02-22:
I forgot to mention this before, but when I try to install the drivers I get a message that says "Homebrew is required" even though it's installed. It won't let me complete the driver portion.

I removed the toolchain and reinstalled and your fix worked! Thanks!

cavearr, 2022-02-22:
Thanks! i'm reviewing homebrew check.

AsynchronousParallel, 2022-02-24:
I'm not having success with a uninstall reinstall latest nightly build Feb10. I update drivers and build has the same error message lib8 not found.

BitBangingBytes, 2022-02-24:
I had to also tell mine to uninstall the toolchain from within icestudio (Tools -> Toolchain -> Remove) and then (Tools -> Toolchain -> Update) for it to work.

Just uninstalling Icestudio and reinstalling it didn't do it.

AsynchronousParallel, 2022-02-24:
Thanks the build worked!!  My upload failed though. ICELink failed to open. iceproghardware.bin and then error.

Obijuan, 2022-04-02:
I will close this issue, as the main problem reported has been solved (libintl.8.dylib. not found)

Please, feel free to open another issues with the other problems (one issue per problem, please). If every issue contains a lot of problems the development team will not be able to track and fix the problems correctly. Thanks

closed 2022-04-02 by Obijuan