FPGAwars/icestudio #619: No internect connection when downloading toolchain

donnmphone, 2022-02-28:


LBertrandDC, 2022-05-02:
Possibly a duplicate of #513 ?

Obijuan, 2022-05-03:
@donnmphone , could you please try the latest wip? You can download it from here:
https://downloads.icestudio.io/

donnmphone, 2022-06-22:
Thx! It doesn't work! It couldn't download the tools chain  !

jojo535275, 2022-06-22:
I had this problem once when my VPN was connected !
So, i f you have one, try disconecting just for the toolchain download/installation!

And if it doesnt do the job, you can try[ installing the toolchain manually](https://github.com/FPGAwars/icestudio/wiki/Installing-the-Toolchains-from-outside-icestudio)!

If you manage to solve the problem on your machine, please let us know whaht worked for you !

cavearr, 2024-02-05:
I will keep the issue open for a few days in case you give me feedback and if you do so later and find it closed, open it without hesitation.

cavearr, 2024-03-18:
I'm closing the issue for not having reply and not be a global problem detected in other users.

If you need help about it , reopen it please!

closed 2024-03-18 by cavearr