FPGAwars/icestudio #624: Blocks menu missing from icestudio

matthewrobe, 2022-03-29:
Hello, I'm having problems with icestudio 0.9.0 on win10 21H2 build 19044:1586 

I'm missing the Blocks section of the menu bar, see screen cap.
[image]
Had the nightly build installed and rolled back to the latest stable release.
does anyone have any ideas?

Many thanks
Matthew

cavearr, 2022-03-29:
Ctrl+T open the new toolbox!  check it and tell us!

matthewrobe, 2022-03-29:
Thanks, Found that :)
[image]
But can't locate the other Block items?

cavearr, 2022-03-29:
the other blocks are on the right , when you could see "Default collection".

Try install the Jedi Collection https://github.com/FPGAwars/Collection-Jedi , that has a lot of blocks.

matthewrobe, 2022-03-29:
Aha! Many thanks, don't know how I missed that :)

Alfredo2591, 2022-04-01:
Tuve que desinstalar la version 9 pues no aparecia el panel basico.
Con la version 3 estable si que aparece.
Actualmente, no me reconoce la Alhambra 2, puede que se deba a tanta reinstalacion.

Obijuan, 2022-04-01:
En la versión 9 la opción del menú básico está en un panel móvil de herramientas (disponible en la opción Editar/toolbox). También se activa con el atajo ctrl-T

Alfredo2591, 2022-04-01:
Ok.Gracias.
No me reconoce la version 3 la Alhambra 2 y eso ....que la tengo en libusbk, actualmente acabo de instalar la 9 y voy a descargar la blockchain.
Anteriormente tenia la camara web, que la he desinstalado.
Espero conseguir que me reconzca la placa.

Alfredo2591, 2022-04-01:
Me encuentro con que a la carga del hola mundo,..me da este error.
[image]

Alfredo2591, 2022-04-01:
Gracias por la información. 🤝👍🙏
No obstante estoy teniendo problemas con la versión nueve, me da error a la
hora de verificar.
Adjunto una captura de pantalla.

[image: Captura de pantalla (13).png]

Obijuan, 2022-04-02:
I will close this issue, as it has been solved

@Alfredo2591 : Tu problema es el mismo que el de @Strauss77

Por favor, sigue las instrucciones indicadas aqui:

https://github.com/FPGAwars/icestudio/issues/494

Esta issue la cerramos porque el panel de Toolchain ya te aparece y sabes cómo sacarlo (no es un bug de icestudio)

closed 2022-04-02 by Obijuan