FPGAwars/icestudio #625: Building errors

ph1lj-6321, 2022-03-31:
[image]

After installing the latest toolchain - I get this error ?

using Win 10 Pc - I've tried various fpga modules - IceStick, BX and Upduino 3.0 - all give same error ?

Alfredo2591, 2022-04-01:
Acabo de instalar la version 9 y ya me pasa eso con el ejemplo basico del Hola Mundo.
Quizas es un problema de la ultima revision.

ph1lj-6321, 2022-04-01:
No, It also appears now on v0.7.0

Philip Jarvis
Professor,
School of Science and Engineering Technology
Electronics Department
Durham College
(905) 721-2000 x2467

[EXTERNAL EMAIL - USE CAUTION]

This message is intended only for the named recipients. This message may contain information that is confidential or exempt from disclosure under applicable law. Any dissemination or copying of this message by anyone other than a named recipient is strictly prohibited. If you are not a named recipient or an employee or agent responsible for delivering this message to a named recipient, please notify us immediately, and permanently destroy this message and any copies you may have. Warning: Email may not be secure unless properly encrypted.

ph1lj-6321, 2022-04-01:
Not just the 0.9.0 - I get the same thing with 0.7.0
very anoying

Obijuan, 2022-04-02:
Hi @ph1lj-6321 

It is a bug that has raised in March-28-2022 because the click python package was upgraded to version 0.8.1 automatically. Installing the toolchains in Icestudio after March-28 causes this error

I am working on generating a new icestudio stable version without that bug. In the meantime, please update the toolchain to the latest stable version:

Assuming you have just installed icestudio, you should have apio 0.8.0. Click on Tools/Toolchain/Update (latest stable)

[image]

It will install the latest toolchain. If you go to Tools/Toolchain you should see Apio version 0.8.0.post1

[image]

Now you should be able to build your circuits without problems

It would be great if you could confirm that this Fix works ok for you. I have only tested it on my linux machine

Obijuan, 2022-04-02:
@Alfredo2591 : Por favor, no pongas el mismo problema en tres issues difererenes. Si no, los desarrolladores nos volvemos locos y no podemos solucionarlos. Gracias

Strauss77, 2022-04-02:
Hola , yo soy un novato de 60 años y he visto el problema y trabajo en Windows 10, al instalar me funcionó todo, pero reinstalé por que en el menú no me aparece el punto "Básico" y me apareció el error, he seguido los pasos y si me ha funcionado, te puedo confirmar que el error de carga desaparece reinstalando la Apio , aunque sigue sin aparecerme en el menu de Icestudio el punto para introducir las entradas y/o salidas
[image]

ph1lj-6321, 2022-04-02:
Alas no – did what you suggested – what I now have can be seen in the enclosed images

Using Win 10

Phil

[icestudio-bug-fixed-01]<https://user-images.githubusercontent.com/1286062/161366854-6ae2bafd-60cd-45ea-aca3-12614fd2643b.png>

[icestudio-bug-fixed-02]<https://user-images.githubusercontent.com/1286062/161366905-a568080f-6138-49ad-aef2-0259ed75b640.png>

ph1lj-6321, 2022-04-02:
Alas no – did what you suggested – what I now have can be seen in the enclosed images

Using Win 1
[image]
[image]
0

Alfredo2591, 2022-04-02:
He seguido los consejos de Obijuan a STRAUSS77   ...... VEASE.....  debug information (#494)
y se solucionaron todos los problemas... (panel basico y error module click)
Gracias Obijuan.

Strauss77, 2022-04-02:
He seguido los pasos que me indica Obijuan y se ha solucionado todo, muchas gracias, ya puedo seguir, me habia bloqueado el ánimo pero ya funciona todo como debería, gracias a todos por el interés que os habeis tomado. un saludo

Obijuan, 2022-04-02:
> 

Hi @ph1lj-6321 

You have installed the toolchain 0.8.0. Now you have to upgrade it

Click on the update (Latest stable) menu option. The new toolchain will be installed. You can check that it has been successfully installed if you see "Apio 0.8.0.post1"

Then, it will work ok (other users have confirm that this solution works for windows)

Obijuan, 2022-04-02:
> 

Genial! Muchas gracias @Strauss77 😃

ph1lj-6321, 2022-04-02:
Yes, I realize that - but every time I do that it just gives me Apio  0.8.0 on one of my machines.

On one machine it did work (Apio 0.8.0.post1- but on the one I need it to work on, it does NOT upgrade

Obijuan, 2022-04-03:
ok. So, I will close this issue as the problem is fixed once the toolchain is updated (apio 0.8.0.post1 fixes it).
The second problems is different. For whatever the reason, in one of your windows machine the toolchain is not correctly updated

I've released a new Icestudio stable version (0.9.0-1) that installs apio 0.8.0.post1 by default. Please, Install this new version and check if apio 0.8.0.post1 is correctly installed

If you find any problem, please fell free to open a new issue with all the possible details (It is important the developers to be able to reproduce the bug reported. We need to isoletate the bugs (in different issues), and collect as much information as possible)

Thanks for you feedback

closed 2022-04-03 by Obijuan

ph1lj-6321, 2022-04-03:
Ok - yes that worked with 0.9.1

cheers (keep up the good work)

Philip Jarvis
Professor,
School of Science and Engineering Technology
Electronics Department
Durham College
(905) 721-2000 x2467

Obijuan, 2022-04-03:
Great!
Happy to hear that 😃