FPGAwars/icestudio #626: Drag output to input

jvahe, 2022-04-08:
Hello everyone,

I'd like to make a feature suggestion: rather than dragging an output to an input on blocks, drag an output to an input.

The reason for this is simple: Many inputs can be connected to a single output, but only one output can be connected to any given input. Basically, the behavior of the editor would behave more like the following game:
https://www.nandgame.com/

I found this to be a bit more intuitive.

In Icestudio, I had to look up how to connect multiple inputs to an output, as nothing was obvious, and found issue #428 of someone having the same issue. Once I did find out how to do it, I found it quite awkward.

Let me know what you guys think of this suggestion. Thanks for hearing me out.

Franck78, 2023-04-12:
Real enhancement would be not reinvent the wheel by rewriting 'another' editor ;)
But rather think how to extend KiCad editor for example ?

cavearr, 2025-01-30:
@lmcapacho do you want to take this?

lmcapacho, 2025-01-30:
Sure, @cavearr  I'll take it

cavearr, 2025-01-30:
Excellent! I am working on the output of the Verilog code that is necessary for other new things and preparing some specifications for the IO manager of the blocks, but if in the meantime you can close these small issues that have been hanging there for a long time that would be great.

Zero issues for the end of February ;)

Thank you very much for joining the iceStudio ship.

jojo535275, 2025-01-30:
Hello guys, i not sure that this issue is correctly explained here ?

So to be sure we are speaking the same:

We can already connect the outputs to the inputs !
What we can not do is connect an input to an output!

Ideally, it will be nice to have both possibilities with of course an aborted connection when we try to connect multiple out to a same input! (we already have this detection)

cavearr, 2025-01-30:
Hi @jojo535275 !!! yes! the idea is that you could connect in both directions,  i  think this was a very good idea but the priorities  do this issue fall in the deep, but i think this will be a very hught improvement in usability.

I'm sendindg all of the new features to the **discussions->ideas** the next days i'll try to organize it to discuss all together. the concept is that issues will only contains bugs and tasks in development process.

I left some *new features* like this, whitout move to *discussions*, because i think could be the moment to put on the table 🥰 

By the way....do you feel like lending a hand with some improvements to the serial terminal? I'm finalizing specifications for several things and since you already lent a hand a while ago, if you feel like it (there's no rush or pressure), tell me and I assign it to you, but only if you feel like it, I don't want to take away your time for other things. And if you're short on time but would like to help with something without having a big task on your shoulders, I can give you some guidance and give you specific tasks to do, just tell me, okay?

jojo535275, 2025-01-30:
Hello Charli,
It was long time ago. But yes i can find some time to play again with the
serial monitor !
Send me a list of your changes ideas! Or assign me the tasks. And i will
have a look !
have a nice evening

cavearr, 2025-01-30:
Give me the weekend to organize all the things! tell you soon! :)

lmcapacho, 2025-01-30:
@cavearr  I support the goal of having zero issues by the end of February. The team is doing great!

I’m interested in contributing to the development of the simulation feature. Please feel free to assign me tasks related to this. Additionally, I’ve been working on the IO changes in the Code block. I’ve made good progress, but I’m currently waiting for the specifications to align my work with the roadmap you’ve defined.

@jojo535275  thank you for clarifying the issue. It’s essential that we’re all on the same page, and I had the same doubt. Your explanation aligns with how I understood it as well.

cavearr, 2025-01-30:
Thank you very much @lmcapacho! I'm counting on you for the simulation function, my idea is to first solve some small problems while I prepare the plan. I want us to have icestudio very stabilized and with some "basic things working" like these last ones that you are solving.

I'm making a new plugin to launch gtkwave and use to document for you and others the plugin system (that should evolve in the coming months, but we'll do it together). From there we have a long but fun road, I will tell you the system that I have in mind but I hope you put all your good ideas and thoughts on the table, my idea is that the specifications are open to define first among everyone and then do it together . In the simulation part the idea is that just as in icestudio we define a circuit with blocks, we can also do it for the simulation, and that flow manager generates the testbench and runs it and we can directly see the result with the gtkwave u another viewer. And on the other hand, I have a topic of virtual simulators underway that will be closely associated with this and that could be very interesting for generating hybrid testbeds.

My goal is to work together in the same direction, avoid collisions and turn iceStudio into a really useful EDA beast 😄

lmcapacho, 2025-01-30:
Thank you @cavearr! All the ideas sound great, and I’m fully on board. I’m here to help in any way I can. Let’s work together to make iceStudio a powerful EDA tool! 🚀

TimRudy, 2025-01-31:
Hi @jojo535275 and @cavearr,

Noticed that there is icerok external vendor file:
- icestudio\app\resources\plugins\icerok\js\vendor\sweetalert2.min.js
It may or may not be the same as sweetalert2.min.js here:
- icestudio\app\node_modules\sweetalert2\dist
(A smaller version of sweetalert2.all.min.js)

Maybe you can point to the latter file, which is available in the project anyway, and remove the former file from 'vendor'. Cheers

cavearr, 2025-01-31:
Hi @TimRudy ! forget icerok, i'm doing a new version totally new , in short i remove this old one and replace for the new, i don't. remove because there are some tutorial with it and while i don't have the first new version  i prefer don't touch.

But don't spent time on this.

TimRudy, 2025-01-31:
👍   *When you remove it, please remove from .prettierignore:
icerok/vendor/
icerok/css/...
(!)