FPGAwars/icestudio #630: Icestudio: Board not connected

MeindlD, 2022-05-23:
Hi Community,

I'm new to the topic of FPGA and Icestudio seems to be a good starting point for me.
So I purchased a iCE40 UltraPlus Breakout Board and downloaded Icestudio on my Windows 10 Client.
I installed Icestudio, the toolchain and the drivers as described in the "Quick Start" - Documentation.
But whenever I try to upload a file to the board I get the Message "Board iCE40 UltraPlus Breakout Board not connected".
J6 is in the position "PROG ICE", J7, J51 and J28 are closed, J11 is open (refer to https://www.latticesemi.com/Products/DevelopmentBoardsAndKits/iCE40UltraPlusBreakoutBoard).
I already changed the Notebook and tried a different breakout board, but the error message stays the same.
I have no idea, what else I could try.

Greetings,
Dominik

Obijuan, 2022-05-25:
Now that I've manage to get access to a windows machine, I will take some more screenshots. This screenshot correspond to the driver installation (Tools/drivers/Enable). If the usb cable is ok, and you have passed the previous checks, connect the board. You will find the device "Lattice ICE40UP5K Breakout (Interface 0)"  (IT IS IMPORTANT TO BE THE Interface 0).  Select the libusK and click on Replace driver:

[image]

It will take some time for windows to install the driver

After that you will see this window:

[image]

Close the window. Now unplug the board and plug it again

Obijuan, 2022-05-25:
This is the board with the previous circuit loaded (I've put some papers on the RGB led)

[image]

In this picture you can see the configuration of my jumpers:

[image]

> **Note**
> The J6 should be in Flash position

But before all of this, check first your USB cable

MeindlD, 2022-05-25:
Hey Obijuan,
first of all: Thanks for the detailed instructions!
So I removed the installed drivers and start again from the beginning to provide you the requested screenshots.
When I plug in the board, the following devices appear:
[image]

So it seems, that my cable is ok (I changed it anyway). The device ID is the same as yours (0403/6010).
But when I try to install the driver, i experience a different behavior: My device is named USB<-> Serial converter
Do i have to install any software from Lattice in advance? The USB ID and the driver versions are the same as yours.
[image]

The driver installation succeeds:
[image]

After plugging the Board out an in again, the following devices appear:
[image]

When I now try to upload your example, i get the error message again:
[image]

The configuration of the jumpers should be the same:
[image]

Do you have any idea, how to proceed?

Thanks in advance,
Dominik

Obijuan, 2022-05-25:
ok, I see. As I am not a windows user/developer I really cannot help much. For whatever the reason your windows system cannot read the board information (it is not related to Icestudio, as Zadig (the driver installer) cannot see it either). So, the first step is trying to make Zadig see it

Maybe you could try to uninstall the USB serial converted A and try Zadig again. Does Zadig see it now? You should read "Lattice ICE40UP5K Breakout (interface 0)
Also try to unplug all the USB devices you can, just in case there is a kind of "interference" between them (who knows, some installed driver that is bothering the others...)
It will try it in Windows again and get more screenshots. I think I do not have the USB-Serial converter drivers... but I am not sure (Making drivers work in Windows is like magic. If you apply the correct spell it will work....)

MeindlD, 2022-05-25:
I tried to remove the driver in several ways now:
First i uninstalled the device and activated the checkbox "Delete the driver software for this device".
[image]

Second i manually deleted the driver files listed for this device:
[image]

But when i disconnect the board and connect it again, the files appear and the driver is installed like it had never been deleted.

Additionally, i downloaded and installed the "Diamond Programmer" from Lattice with its drivers. Also no success.
The same drivers as before are installed, when i plug in the board.

This drives me nuts...

TimRudy, 2022-05-25:
Hi Meind,

I had the same issue (FPGAwars issue #234), and tried all kinds of
software/driver and Zadig/install attempts, but it isn't a software issue,
it is an issue where you need to give Windows a kick by using an actual new
USB port or ID (that it assigns). I guess you already tried all the USB
ports on your machine? Now, to put a new USB hardware ID in its USB
hierarchy all you have to do is borrow a USB hub - bus-powered - for
example a 1-to-4 hub. (Another example is I had a remote USB port on a
monitor, and that worked.) With different hardware, you should get
different result. Windows can need a bit of magic (or give you a bit of
magic) like Obijuan said. Good luck!
  Tim

MeindlD, 2022-05-25:
Hi Tim,
thanks for joining the discussion 😂
It's good to hear, that I am not the only person in the world with this problem.
According to what you proposed, i used the USB Hub in my monitor, aaaaaaaaaaaaand...

at least the error message changed 🤣
[image]

Do you know if this is a related issue or a different topic?

TimRudy, 2022-05-25:
Different issue, getting APIO working. The Windows issue was  
[image]
or  
[image]

Obijuan, 2022-05-25:
These errors are also related to the drivers. Usually they show up when the driver is not installed (the software tries to open the device but it receives an error from the driver). The typicall action is to install the driver

After you get the error, can you click on the View/Command output and send an screenshot? I need to collect more information about it

(I assume that now Zadig worked ok and you already got the ICE40UP5K Breakout (Interface 0) Label)

MeindlD, 2022-05-30:
Hi Tim and Obijuan,

thanks again for your support. Sorry for the late answer, but I didn't have access tot he hardware the past few days.

So this is my Command Output:
[image]

In the device manager, my board is still named "USB <-> Serial Converter". But as I already wrote above: I have no idea, how i could get rid of this driver. Nevertheless the right driver for the board should have been installed, as it is listed as "libusbK USB devices"
[image]

zambujal, 2023-06-05:
don't know if it's the same thing... but I had some problems with the icestick lattice just changed the usb descriptor name... 

The descriptor doesn't correspond (they changed it) so you have to change the boards.json (search for it in the users folder...) to the new usb descriptor for this to work.
[https://blog.j2i.net/2023/01/27/resolving-board-icestick-not-connected-for-the-lattice-icestick-hx1k-with-apio/](url) Apio as the same problem.

holography-be, 2023-07-06:
Hello,
I have the same problem with my icestick :-(

[image]

Where did you found the boards.json ?
I'm looking on my all disk but not found it.

Thank you.

cavearr, 2024-03-18:
The last wip (https://downloads.icestudio.io) contains the development toolchain with the latests yosys version that is more compatible with usb in windows (now you don't need zadig).

Try it and if you have problems reopen this.

closed 2024-03-18 by cavearr