FPGAwars/icestudio #633: "Error executting command" at first run

denizzzka, 2022-06-03:
Hi! I am newbie

Tried to run both RC and WIP AppImages on clean Debian Linux system, but after selection of any board got this error during `virtualenv` creation:

[image]

Python 3.10 is installed on my system:
```
denizzz@denpc:~$ ls -l /usr/bin/python3.10
-rwxr-xr-x 1 root root 5896776 May 14 08:40 /usr/bin/python3.10
```

cavearr, 2022-06-03:
Try installing distutils:

sudo apt install python3-distutils

denizzzka, 2022-06-03:
> Try installing distutils

Already installed, with `-extras`

Also found in console:
```
/bin/sh: 1: /usr/local/Cellar/python/3.8.2/bin/python3: not found
/bin/sh: 1: /usr/local/Cellar/python/3.7.7/bin/python3: not found
```

Obijuan, 2022-06-04:
Hi @denizzzka ,

Please, check the instalation instructions on the wiki

https://github.com/FPGAwars/icestudio/wiki/Installation#ubuntu-2004

These instructiins are for Ubuntu, but should also work for Debian

It seems that you have missed to install the python3-venv package

The error message you get is raised because the virtual python env (venv) cannot be executed

denizzzka, 2022-06-04:
@Obijuan thank you! Yes, I skiped this step. Now it works!

closed 2022-06-04 by denizzzka

darkredhorse, 2022-10-30:
Although issue is closed, I thought I would just add a confirmation that the windows installer also had an issue, but I just rand through the manual install steps and it seem s good to go.   Repeated warnings about the platform name "windows_amd64" not matching popped up, but the scripts then attempted "windows" and exited successfully.