FPGAwars/icestudio #634: New policy for testbench ?

laulin, 2022-06-03:
Hello :) I just installed the 0.10-rc1 and I was very surpised : 

- Export ot verilog is maked as obsolete
- Export of testbench produces non usable TB (previously, ok with some minors changes).

What happened ? Why ? Do I miss something about a new way to do test of design ?  

Design complex stuffs without TB is a little bit tricky ...

cavearr, 2022-06-03:
Hi!, we are changing a lot of things in the last wips and releases. If you want to try the last wips you could donwload from https://downloads.icestudio.io

The verilog export is marked as obsolete because now, in the same folder that you have your .ice file, when you synthetize it, icestudio create a new folder name ice-build.

Inside it you could find the main.v file with the verilog output of your design.

About the test benches could be a bug in the last versions, i'm looking for it to understand what is it happening with it.

Thanks!

laulin, 2022-06-04:
Ok, I figure out how to create TB and validate my design (awsome !). As usual, the problem was between the chair and the keyboard.

Now it works for me, How can I write a tutorial/FAQ/howto/wiki/doc about basic stuff to test a design ?

Obijuan, 2022-06-04:
Hi Laulin!

I've invited you to the wiki team.  Once the invitation is accepted, please check if you can edit the wiki page (I am not sure if I've set the correct privileges)

If you can edit, feel free to add any documentation you like

laulin, 2022-06-05:
HI Obi, Thank you for your confidence ! 

I don't see "edit" nor "new" buttons in wiki page so I think I don't have rights.

Obijuan, 2022-06-05:
Could you please try it now? (If it still does not work I will try something different. I am still learning about github teams)

laulin, 2022-06-05:
Yes it works :) Do you have a _place_ (discord, forum, etc) outside of issues to talk about icestudio ?

Obijuan, 2022-06-05:
yes, we have the following google group:

https://groups.google.com/g/fpga-wars-explorando-el-lado-libre

Initially was only in Spanish but now you could write in English. You are very welcome to join and comment/discuss your ideas