FPGAwars/icestudio #635: Tips about SVG image

laulin, 2022-06-03:
Hello :) 

Do you have tips to share about having good component image (icon ?) ? What piece of software to you use to create them easily and fastly ?

cavearr, 2022-06-03:
Hi Laulin! you can use inkscape , it is open source and multi platform.

https://inkscape.org/es/

laulin, 2022-06-04:
Indeed, inkscape is an exellent vector drawing tool. But it was not my mind, I was more thinking about stuffs like that : 

- https://suncatcherstudio.com/image-text-editor/ : create a SVG based on a text, good to create fast label for you components
- https://www.flaticon.com/fr/ : find icon in svg 

May be some other users have that kind of tips ? Do you think such links can be added to the wiki ?

cavearr, 2022-06-05:
Ah! i don't understand you initialiy! i'm look for icons usually at https://www.iconfinder.com/
But there are many open source sites, i'll try to put in this thread or in the wiki.

Thanks!

cavearr, 2023-06-29:
For the moment , i'm closing this thread, is you need clarify something reopen it.

closed 2023-06-29 by cavearr