FPGAwars/icestudio #636: Uploading fails without error message

Pasca133, 2022-06-04:
I am currently working with an icebreaker board. On my main pc (windows 10) it works fine. On my laptop (also windows 10) it fails without an error. The verification an building process works fine. When I open the command output to debug it sadly doesn't show any error message. It just quits after the debug lines. The output looks like this: 

set APIO_HOME_DIR="C:\Users\Pascal\.icestudio\apio"& set PATH=C:\Users\Pascal\.icestudio\venv\Scripts;C:\Program Files\Python310\Scripts\;C:\Program Files\Python310\;C:\Windows\System32;C:\Windows;C:\Windows\System32\wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Intel\iCLS Client\;C:\Program Files (x86)\Intel\iCLS Client\;C:\WinAVR-20100110\bin;C:\WinAVR-20100110\utils\bin;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Program Files (x86)\Wireless Docking\;C:\Program Files\MiKTeX 2.9\miktex\bin\x64\;C:\Users\Pascal\AppData\Local\Programs\Python\Python39\Scripts\;C:\Users\Pascal\AppData\Local\Programs\Python\Python39\;C:\Users\Pascal\AppData\Local\Microsoft\WindowsApps& "C:\Users\Pascal\.icestudio\venv\Scripts\apio.exe" upload --board iCEBreaker -p "C:\Program Files\icestudio\resources\collection\examples\1. Basic\ice-build\06. Pushbutton and LED"

(DEBUG) Profile path: C:\Users\Pascal\.icestudio\apio\profile.json
(DEBUG) Home_dir: C:\Users\Pascal\.icestudio\apio
(DEBUG) Profile path: C:\Users\Pascal\.icestudio\apio\profile.json
(DEBUG) Home_dir: C:\Users\Pascal\.icestudio\apio
(DEBUG) Run Command: lsusb
(DEBUG) System_base_dir: C:\Users\Pascal\.icestudio\apio\packages\tools-oss-cad-suite
(DEBUG) System bin dir: C:\Users\Pascal\.icestudio\apio\packages\tools-oss-cad-suite\bin
(DEBUG) Executable file: C:\Users\Pascal\.icestudio\apio\packages\tools-oss-cad-suite\bin\lsusb.exe

It fails even before it checks if a board is connected. So the "error" comes up even if no board is connected. 
I reinstalled python, icestudio (tryed different versions of both), tried to clean up my PATH und User PATH, but nothing worked so far. The next step would be to reinstall windows, but I hope it wont be necessary. 
Anyone an idea what the problem might be?
A little bit hard to debug without an error message.

Obijuan, 2022-06-04:
It seems that a wrong path for the icestudio profile is being used. Let's check what is in your system. Cold you please check if the folder c:\Users\Pascal exist? I assume it should be your home user folder. If it exists, could you provide an screenshot of the file manager showing the files in that folder?

(For unknowkn reasons, your icestudio profile folder path is c:\Users\Pascal.icestudio, but it seem to be wrong. It should be C:\Users\Pascal\.icestudio. But I need to collect more information on your system for knowing what's going on)

Pasca133, 2022-06-04:
[image]
yes it exists. 
I just realised the path for icestudio is C:\Users\Pascal.icestudio\ but it should be C:\Users\Pascal\.icestudio\ with a \ before the .icestudio, shouldn't it?

Obijuan, 2022-06-04:
ok, yes, C:\Users\Pascal exists, and the correct .icestudio folder is there, so that is correct.  I am thinking that maybe the missing \ is also scaped in the console view, and for that reason it is not shown (in that case the error is not because of that)

Let's check that all the folders are ok. I need you to attach another screenshot of the Icestudio window that is shown when clicking on the View/System Info menu option

(I think it does make any difference to use "C" or "c")

Pasca133, 2022-06-05:
[image]

Obijuan, 2022-06-05:
Sorry, I did not explain well myself. What I wanted you to do is: Open Icestudio, click on the View/system info menu option and take a screenshot of the new panel that will show up. It should looks like this:

[image]

Pasca133, 2022-06-05:
[image]
[image]

Obijuan, 2022-06-05:
Thanks Pascal. I can now confirm that all the forlders are ok. So, the problem is due to something not related with the folders.

Let's try to find more information. These are the step I want you to follow:

1) Close Icestudio
2) Remove the file C:\Users\Pascal\icestudio.log
3) Open Icestudio
4) Open the example that is not working and upload it (this is for reproducting the problem)
5) Close Icestudio
6) Attach the icestudio.log (this file if automatically generated if it did not exist before)

The Icestudio.log file is a text file that shows more information. Let's keep our finger crossed, hoping to see some error messages that could give us some clues about the problem

Pasca133, 2022-06-05:
[icestudio.log](https://github.com/FPGAwars/icestudio/files/8839701/icestudio.log)

No Error message :/

Obijuan, 2022-06-05:
ok, but the icestudio.log file seems incomplete. I would like to see it with the full information.  Please, follow these sptes (do not delete the current icestudio.log)

1) Close icestudio (if it was already running)
2) Run Icestudio (it will add some information to the end of the current log)
3) Try the design
4) Close icestudio
5) Attach the icestudio.log

I want to see the full file, because It looks strange

This is an example of my icestudio.log file, generated uploading a design to the icebreaker board:

[icestudio.log](https://github.com/FPGAwars/icestudio/files/8839730/icestudio.log)

Pasca133, 2022-06-05:
[icestudio.log](https://github.com/FPGAwars/icestudio/files/8839801/icestudio.log)

Obijuan, 2022-06-05:
ok. Everything seems to be all-right. This a kind of "x-file"

The only way to know what is happening is to use Apio to directly upload files into the icebreaker board (Icestudio calls apio for building/uploading).  I think the problem is related to the installed windows driver

Apio is installed form the command line (cmd).The [full instrucctions are here](https://github.com/FPGAwars/apio/wiki/Quick-start#windows-10), but I will sumarize it

Please follow these instructions and provide all the information you can from every step

1) Open the cmd line in windows
2) Check the python version: execute the command "python --version"
(it should be python 3.9 or higher)
3) Execute the command: "pip install apio"  (It will install apio in your system)
4) Excute the command: "apio install -a" (It will install the toolchain for synthesizing/uploading bitstreams)
5) Now install the driver and perform some test. Follow these instrucctions: https://github.com/FPGAwars/apio/wiki/Quick-start#testing-the-icebreaker-board-1

It is specially import the part of the driver installation. It would be great if you could provide screenshots. Does Zadig detect your iceBreaker board? (Does it show the same names that you can see on the wiki screenshots) (I need to know what is different in your laptop, to get some glues about the problem)

Pasca133, 2022-06-05:
C:\Users\Pascal>python --version
Python 3.10.4

C:\Users\Pascal>pip install apio
Defaulting to user installation because normal site-packages is not writeable
Collecting apio
  Using cached apio-0.8.1-py3-none-any.whl (70 kB)
Collecting pyserial==3.5
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting scons==4.2.0
  Using cached SCons-4.2.0-py3-none-any.whl (4.2 MB)
Collecting colorama==0.4.4
  Using cached colorama-0.4.4-py2.py3-none-any.whl (16 kB)
Collecting wheel<1,>=0.35.0
  Using cached wheel-0.37.1-py2.py3-none-any.whl (35 kB)
Collecting click==7.1.2
  Using cached click-7.1.2-py2.py3-none-any.whl (82 kB)
Collecting semantic_version==2.9.0
  Using cached semantic_version-2.9.0-py2.py3-none-any.whl (15 kB)
Collecting requests==2.27.1
  Using cached requests-2.27.1-py2.py3-none-any.whl (63 kB)
Collecting certifi>=2017.4.17
  Using cached certifi-2022.5.18.1-py3-none-any.whl (155 kB)
Collecting idna<4,>=2.5
  Using cached idna-3.3-py3-none-any.whl (61 kB)
Collecting charset-normalizer~=2.0.0
  Using cached charset_normalizer-2.0.12-py3-none-any.whl (39 kB)
Collecting urllib3<1.27,>=1.21.1
  Using cached urllib3-1.26.9-py2.py3-none-any.whl (138 kB)
Requirement already satisfied: setuptools in c:\program files\python310\lib\site-packages (from scons==4.2.0->apio) (58.1.0)
Installing collected packages: pyserial, wheel, urllib3, semantic_version, scons, idna, colorama, click, charset-normalizer, certifi, requests, apio
  WARNING: The scripts pyserial-miniterm.exe and pyserial-ports.exe are installed in 'C:\Users\Pascal\AppData\Roaming\Python\Python310\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script wheel.exe is installed in 'C:\Users\Pascal\AppData\Roaming\Python\Python310\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The scripts scons-configure-cache.exe, scons.exe and sconsign.exe are installed in 'C:\Users\Pascal\AppData\Roaming\Python\Python310\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script normalizer.exe is installed in 'C:\Users\Pascal\AppData\Roaming\Python\Python310\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script apio.exe is installed in 'C:\Users\Pascal\AppData\Roaming\Python\Python310\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed apio-0.8.1 certifi-2022.5.18.1 charset-normalizer-2.0.12 click-7.1.2 colorama-0.4.4 idna-3.3 pyserial-3.5 requests-2.27.1 scons-4.2.0 semantic_version-2.9.0 urllib3-1.26.9 wheel-0.37.1
WARNING: You are using pip version 22.0.4; however, version 22.1.2 is available.
You should consider upgrading via the 'C:\Program Files\Python310\python.exe -m pip install --upgrade pip' command.

(I upgraded pip really quick, but nothing special there. What confuses me is the fact that python 3.10 is still installed. I uninstalledd everything this morning and used the 3.9 version which comes with the icestudio installation.)

Pasca133, 2022-06-05:
C:\Users\Pascal>C:\Users\Pascal\AppData\Roaming\Python\Python310\Scripts\apio.exe install -a
(DEBUG) Profile path: C:\Users\Pascal\.apio\profile.json
(DEBUG) Home_dir: C:\Users\Pascal\.apio
Installing drivers package:
Warning: full platform does not match: windows_amd64
         Trying OS name: windows
Download tools-drivers-windows-1.1.0.tar.gz
Downloading  [####################################]  100%
Package 'drivers' has been successfully installed!
(DEBUG) Profile path: C:\Users\Pascal\.apio\profile.json
(DEBUG) Home_dir: C:\Users\Pascal\.apio
Installing examples package:
Download apio-examples-0.0.33.zip
Downloading  [####################################]  100%
Package 'examples' has been successfully installed!
(DEBUG) Profile path: C:\Users\Pascal\.apio\profile.json
(DEBUG) Home_dir: C:\Users\Pascal\.apio
Installing gtkwave package:
Warning: full platform does not match: windows_amd64
         Trying OS name: windows
Download tool-gtkwave-windows-3.3.77.tar.gz
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Package 'gtkwave' has been successfully installed!
(DEBUG) Profile path: C:\Users\Pascal\.apio\profile.json
(DEBUG) Home_dir: C:\Users\Pascal\.apio
Installing oss-cad-suite package:
Download tools-oss-cad-suite-windows_amd64-0.0.8.tar.gz
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Package 'oss-cad-suite' has been successfully installed!

unfortunately I can't properly test any further today because I don't have the board here. The Error comes up either way (with or without board). Normally it should give the error the board is not connected when I'm starting to upload but it doesn't even come this far so I traveled without the board this weekend trying to get to the "no board connected" error instead of my current error.
From my past testings I can tell you that Zadig dected and changed the drivers just fine. Looks exactly the same like on my main PC.

Obijuan, 2022-06-05:
It should also work with python10, but for the safe side, let's try to use only python9.  Uninstall apio first (use the command "pip uninstall apio"), because it has been installed with python10.  Then try to remove python10. The goal is that the comand "python --version" from the cmd line shows only python 3.9

Once that is ok, let's proceed with the next steps

Obijuan, 2022-06-05:
> unfortunately I can't properly test any further today because I don't have the board here. The Error comes up either way (with or without board). Normally it should give the error the board is not [...]
From my past testings I can tell you that Zadig dected and changed the drivers just fine. Looks exactly the same like on my main PC.

OK. With apio installed and once you have the board, i want you to do these tests:

* Execute the command "apio system --lsusb". If the board is connected, you should see a line similar to these one (among other lines):
```
0403:6010 (bus 1, device 8) path: 5
```

* Execute the comand "apio system --lsftdi" 

```
Number of FTDI devices found: 1
Checking device: 0
Manufacturer: 1BitSquared, Description: iCEBreaker V1.0e
```
Make sure you get that exact information

Pasca133, 2022-06-07:
[image]
[image]
maybe this is the error
By the way: the name of the icebreaker board is a bit odd, but on my main PC it shows exactly the same, so this shouldn't be an issue.

Pasca133, 2022-06-07:
Because I need icestudio to work on my laptop, I reinstalled Windows 10. And it works just like normal now.

closed 2022-06-07 by Pasca133

Pasca133, 2022-06-08:
[image]

Obijuan, 2022-06-08:
Thanks you very much @Pasca133 ! 
🙂️