FPGAwars/icestudio #639: Linux ARM64 builds are actually 32-bit ARM

janrinze, 2022-06-21:
Just tried to install icestudio ARM64 builds () on asahi linux (Apple M1 Mac Mini) but discovered that icestudio is built as a 32 bit application.
```
janrinze@M1DebianSid:~/appsrcs/icestudio-0.10-rc1-aarch64$ file icestudio 
icestudio: ELF 32-bit LSB pie executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.2.0, BuildID[xxHash]=b16e95ba333ec97c, stripped
```

My other ARM systems can run 32 bit alongside 64 bit applications. However the Apple M1 does not support 32 bit ARM binaries.
Will we have real 64 bit ARM builds of icestudio or is there some limitation there?

cavearr, 2022-07-05:
Hi @janrinze! this is in the "todo list" because we dont have asahi linux to test it.
If you could give us feedback in the next days i could generate new wips for arm64 in real64 bits. Could you try it if i compile it?

janrinze, 2022-07-05:
Sure can try.  let me know where to download.

cavearr, 2024-02-05:
Now you have linux arm64 native support bundles in the last wip , you could download from github actions or more easily from https://downloads.icestudio.io

closed 2024-02-05 by cavearr