FPGAwars/icestudio #642: `npm install` fails on M1 Mac

honteng, 2022-06-29:
Running `npm install` fails as follows

```
➜ npm i
npm ERR! code 1
npm ERR! path /Users/honten/workspace/github.com/icestudio/node_modules/nw
npm ERR! command failed
npm ERR! command sh -c /var/folders/f1/n_pgdrbn50nb5j1_w73lfzdr0000gn/T/postinstall-1656480775273.sh
npm ERR! Response code 404 (Not Found)

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/honten/.npm/_logs/2022-06-29T05_32_49_457Z-debug-0.log
```

To resolve issue, we need to add `npm_config_nwjs_process_arch=x64` options as follows

```
npm_config_nwjs_process_arch=x64 npm i
```

I guess we should update README.md. If you want me to do so, I can create PR. Or can you update the doc?

cavearr, 2022-07-05:
Hi @honteng !, Make a PR! other option is use terminal with Roseta .

cavearr, 2024-02-05:
Hi! i'm added osx arm 64 native support, you could donwload from https://downloads.icestudio.io or clone the repository in dev branch.

Thanks!!

closed 2024-02-05 by cavearr