FPGAwars/icestudio #643: Basic FPGA Board: Recommendation

MeindlD, 2022-07-04:
Hi Community,

I'm quite new to the topic and I'm looking for a suitable board for my purposes:
I tried to start with the iCE40 UltraPlus Breakout Board, but couldn't get it running (see issue #630).

I plan to treat this topic with trainees in my company in the future, so a need a reliable an easy-to-use environment.
I'm looking for a simple board which is still in production and available in Germany.

Do you have any recommendations for me?

Greetings,
Dominik

cavearr, 2022-07-04:
Hi @MeindlD !a lot of icestudio users, use alhambra II board, this is produced in Spain but they buy around the world. This is a great board to start.

Check if you want https://alhambrabits.com/buy/

MeindlD, 2022-07-06:
Thank you, cavearr,
the Board look suitable for my purposes.
It's a pity, that there is no real online shop for Germany, where I can buy it.
It could cet difficult to order it (it will be payed by my company).
But thanks anyway!

cavearr, 2022-08-17:
Hi @MeindlD ! what board do you bought finally?

MeindlD, 2022-08-31:
I chose the Alhambra II Board. Shipping from Spain to Germany was a bit tricky.
Today the board arrived and I immediately plugged it in. It worked from the beginning without any issues.

closed 2023-06-29 by cavearr