FPGAwars/icestudio #645: Support for Apicula compatible boards?

Harvie, 2022-07-21:
Hello,
can you please add boards supported by apicula project?
https://github.com/YosysHQ/apicula

Namely:
    Trenz TEC0117: GW1NR-UV9QN881C6/I5
    Sipeed Tang Nano: GW1N-LV1QN48C6/I5
    Sipeed Tang Nano 1K: GW1NZ-LV1QN48C6/I5
    Sipeed Tang Nano 4K: GW1NSR-LV4CQN48PC7/I6
    Sipeed Tang Nano 9K: GW1NR-LV9QN88PC6/I5 
    Seeed RUNBER: GW1N-UV4LQ144C6/I5
    @Disasm honeycomb: GW1NS-UX2CQN48C5/I4

cavearr, 2022-07-29:
Hi @Harvie ! i don't have any of this boards , i'll buy it one of them and prepare the integration along the next weeks.

I'm maintaining this thread open to inform you about the progress.

cavearr, 2022-07-31:
Hi @Harvie ! boards are comming, asap i'll been working on it

Harvie, 2022-08-03:
Thank you! Which one did you ordered? So far i have `Sipeed Tang Nano 9K`, so i can help you with testing...

k0055t, 2022-08-15:
How is the development?
Are there any problems with implementing support? I also have Sipeed Tang Nano 9K, so can help with testing. I would like to program the board with opensource toolchain. Thanks for your work

cavearr, 2022-08-15:
Hi team! i'm waiting for my board, i bought two models:

- Tang Nano  9K 
- Sipeed Lichee Tang Nano GW1N-1

In theory , boards should be arrive one week ago, but for the moment the estimated date to arrive is for August 22.

I'm waiting for it, as soon as i have it, i'm telling you, all your help , tests ,feedback and ideas in the integration process should be very useful.

Thanks a lot !

cavearr, 2022-12-15:
Hi @Harvie! I am very close to the apicula support! Thank you!, in this thread I will notify the initial versions, your comments and tests will be very important.

Thank you and keep this thread to talk about it.

kami4ka, 2023-03-17:
Any news?
The boards get more popular.

Harvie, 2023-11-01:
Fixing this would be great idea for xmas present :-D

cavearr, 2023-11-01:
Hi tomas! With hight probability in christmas this integration will be finished 😉

cavearr, 2024-03-07:
I'm very near to publish the integration. Probably at the end of the month at the latest.

there's little left!

Greeniac1, 2024-07-28:
Hey, any updates? thank you for the work so far

cavearr, 2024-07-28:
Hi @Greeniac1 ! this will be very soon ready to test.  Sorry for the delay but i'm rewriting some core parts needed to this and take me more time that expected. Along  August should be ready. I'll maintain you informe here.

Thanks!

Harvie, 2024-07-29:
@Greeniac1 haha did you came here after Great Scott featured Tang Nano 9K? :-D
https://youtu.be/BcUzuwP-WPI?t=597

Greeniac1, 2024-08-14:
@Harvie  haha, no i actually bought it like a year ago, i just bought the cheapest fpga on aliexpress, and quickly discovered that vertigo is nothing like C or Python (the only 2 languages that i know) and i just didn't learn how to program it. so thats why  i became interested in icestudio.

TheZ3ro, 2024-09-04:
@cavearr is there a way we can help with the development?

cavearr, 2024-09-05:
Hi @TheZ3ro ! sorry for my long delay with this task but i'm working in a big core changes and i need to move the priority of some tasks.  Apicula support is very near to be working, i hope give you good news in the next weeks.

karaketir16, 2025-01-06:
Are there any updates? It looks like a great tool, but I only have Tang Nano.

cavearr, 2025-01-06:
Hi! Yes se finish the apicula integration few days algo and in a very few days we will release apicula support for icestudio.