FPGAwars/icestudio #648: alchitry cu

juanmicro, 2022-08-03:
icestudio tiene un error en el  Pinout  tx el numero de pin corresponde al  de rx  ;

cavearr, 2022-08-17:
Podrías indicarme los números de pin para el tx y el rx?

No tengo esta tarjeta y no puedo validarlo, si pudieras indicármelo, realizo los cambios genero una nueva wip y si pudieras probarla te lo agradecería.

Gracias!

juanmicro, 2022-08-19:
hola yo lo modifique asi y funciona perfectamente
# ------------ FTDI 1 ---------------------------------------------------------
# UART
set_io --warn-no-port TX  M9  # output
set_io --warn-no-port RX  P14  # input

gracias por el programa que habéis creado estoy muy agradecido.

juanmicro, 2022-10-11:
hola yo lo modifique asi y funciona perfectamente
# ------------ FTDI 1 ---------------------------------------------------------
# UART
set_io --warn-no-port TX  M9  # output
set_io --warn-no-port RX  P14  # input

gracias por el programa que habéis creado estoy muy agradecido.

Enviado el: ‎miércoles‎, ‎17‎ de ‎agosto‎ de ‎2022 ‎23‎:‎02
Para: ***@***.***>
CC: c ***@***.***>, ***@***.***>

cavearr, 2024-02-07:
Sorry for the long delay, your changes has been applied and are available in the next hours in the latest wips . 

You can update from github actions actifacts or easily from https://downloads.icestudio.io

closed 2024-02-07 by cavearr