FPGAwars/icestudio #649: Performance Issues Icestudio

BPJWES, 2022-08-21:
When working on a large project for the ULX3S, Icestudio becomes very laggy.

After connecting a wire, the user interface (drawing canvas) locks for up to 10 seconds.

0.9.2.w2022907250707
running on windows 11
AMD Ryzen 9 5900HX

[ICE_FILE_DEMONSTRATING_PERFORMANCE_ISSUES.zip](https://github.com/FPGAwars/icestudio/files/9389449/ICE_FILE_DEMONSTRATING_PERFORMANCE_ISSUES.zip)

cavearr, 2022-08-21:
Hi BPJWES! thanks for your feedback, i'm going to research around your example.  We are working in a lot of improvements but i'm looking if in you design are something that generate an internal bug .

Tell you this days

cavearr, 2024-02-05:
I'm closing this issue because the performance is increasing in the last wips https://downloads.icestudio.io

Reopen it if you thinks it is needed.