FPGAwars/icestudio #650: ICEFun board, LEDs no reaction

FPGA7, 2022-08-27:
I worked with TinyFPGA BX, all ok. 
Then I changed to ICEFun board. 
[image]
Form me there's no way on switching the selected LED on. But when choosing speaker as output I can hear the prescaled clock signal as wanted. Is there any special point to respect with LEDs on this board?

cavearr, 2022-08-27:
I think this is a problem for the pcf file. This is the description of the board.
I don't have the icefun board, if i'll try some changes could you help me in the debugging process?

Thanks!

FPGA7, 2022-08-27:
Thank you for your help offer. I will help you in the debugging process.

FPGA7, 2022-08-28:
I've got it, a very simple mistake. ICEfun has an array og LEDs. Each LED is defined by line and number. So you have to declare in addition the line.
[image]
.

closed 2022-08-28 by FPGA7

cavearr, 2022-08-29:
Thanks for the feedback! i'm buying an ICEFun for future problems.

Thanks again!