FPGAwars/icestudio #651: After exporting subblock, all "block interfaces" are broken

BPJWES, 2022-09-01:
After exporting the following block from my design:
[image]
Using save as.

When I reopen that block, all the inputs and outputs of the block, (colored grey above), are broken (at least graphically), see below.

(On windows 11)

[image]

BPJWES, 2022-09-01:
Here a minimal demonstration of the bug is added:
[Demonstration of Bug.zip](https://github.com/FPGAwars/icestudio/files/9468550/Demonstration.of.Bug.zip)

This file contains 16bitSplitter.ice.

If you open a new empty project in ICEstudio, add as block 16bitSplitter.ice. Then open that block ('enter the design') and save as, you will get the result as added in TESTJE_FOR_BUG.ice

In which all the ports have been mangled

cavearr, 2022-09-01:
Hi @BPJWES! when you save a module as .ice you need import "as block":

[image]

This is as is, because when you open icestudio in the "top level", inputs and outputs need to be physical and in your .ice don't exists.

Try to open "as block" and tell us if you see a new block with your design in it.

BPJWES, 2022-09-01:
I open the saved (exported) block stand alone on purpose, because I want to edit it for later reuse in different projects. Therefore I think this top level issue should not occur. Because this issue limits the hierarchical design method which makes Icestudio so nice in my opinion.

In my humble opinion, the following should be valid.

Design a building block, for example, building_block_1.ice, and save it.

Then make a design with this building block (using the add as block function), for example, design1.ice .

Then if you enter the block (building_block_1), from within design1.ice, and you have not edited it, and "save as" (thereby exporting the block) building_block_2.ice

In this scenario I think building_block_1.ice and building_block_2.ice should be identical.

 

_Try to open "as block" and tell us if you see a new block with your design in it._
This works fine, but I want to specifically edit the "building blocks".

cavearr, 2024-12-21:
@TimRudy @BPJWES this is fixied in the last wip with development toolchain. If you try it and find new errors or persist reopen it, plase.

Thanks!

closed 2024-12-21 by cavearr