FPGAwars/icestudio #652: Guardar como:

Democrito, 2022-09-03:
Si un submódulo queremos guardarlo aparte, cuando le damos a "Guardar como:" y queremos utilizarlo, no sintetiza y no hay mensaje de error, sin embargo su aspecto cambia levemente:
El suavizado de las cajas (modulos e I/O)  se pierde y pasa todo a ser rectangular, dejo dos imágenes para contrastar visulamente.

[image]

[image]

jojo535275, 2022-09-04:
Hola Democrito, 
Your two images above are identical !?

Democrito, 2022-09-04:
They are "almost" identical. Look at the corners of the elements, in one image they are rounded and in the other image they are square. When I see that they are square I know that he will not be able to synthesize.

This only happens when we try to save a submodule for use in another project.

And it's not important, but I'll leave the warning so you know what's going on.

Cheers!

jojo535275, 2022-09-04:
ah ok, i see now !
Thanks for that info!

As i am playing with your log blocks, I just tried to reproduce that error with your "serial_example_ln.ice" file ( "saving as" the "Ln___Q16_16"),
Then opening it a new ice file, and i could "import as a block" the Ln___Q16_16.ice file without the above problem!?

Do you see it every time, with every of your ice files, when you try "saving as a block" and then importing that saved file as a block

Big hug!

Democrito, 2022-09-04:
Oops! I've followed my own steps (twice) and now I don't get that error. What I mentioned above happened to me in the past and yesterday it happened to me again. That's why I thought it always happened. If it happens again I will send you the example.

A big hug!

cavearr, 2022-09-05:
Hi team! i'll try to review this strange behaviour, thanks for reporting! if you find some pattern that generate the error always, please report it!

cavearr, 2024-12-21:
@Democrito @jojo535275 i think this issue is fixed in the last wips , could you confirm to close this?

Democrito, 2024-12-22:
That's right, this issue was resolved a few Wips ago. I'm closing the issue.

closed 2024-12-22 by Democrito