FPGAwars/icestudio #656: cannot be legalized: initialized dlatch are not supported

bogatkaj, 2022-10-23:
Hey there,

I have a problem compiling the attached project after adding a second case in the code block.
At compiling it give this error:
There are errors in the Design...
`ERROR: FF main.$auto$simplemap.cc:581:simplemap_dlatch$877 (type $_DLATCH_N_) cannot be legalized: initialized dlatch are not supported`

I guess I missed something but the "FF" in combination of the usage of the RS flip-flop of the iceFF-0.1 irrtates me.

I hope someone can help me or clarify this as bug.
[Test_DIP_7SEG.ice.tar.gz](https://github.com/FPGAwars/icestudio/files/9846756/Test_DIP_7SEG.ice.tar.gz)

TimRudy, 2024-01-19:
Strange as it seems, that is the synthesizer's way of stating this:  
&nbsp;&nbsp;&nbsp; "You did not put an `else` at the end of every `if` statement, even one simple `if` statement, or an `if` statement followed by any chain of `else if` statements, to close off the input space of possibilities"  
AND  
&nbsp;&nbsp;&nbsp; "You did not put a `default` at the end of every `case` statement, thus ensuring it has full coverage of its input values"  

Here's a quick description that you can find better documented elsewhere:

Hardware generated by Verilog is very open-ended, but you have to follow some simple rules like that to keep it under control. For these conditions (branches), if all the possible data values are not "covered" to describe & derive your outputs, then it believes you are leaving an input value as it was till next transition. To do that, the synthesizer creates a latch. Why is it telling you you didn't want the latch? I'm not quite sure, but consider yourself lucky that it stopped you...?

I think everyone can agree the error messages are awful, sometimes. Not much guidance...

But for Icestudio to add a validation layer to check your `if`s and `case`s before it sends the code down to Yosys... I'm guessing that might be taking on too much responsibility from Yosys, and opening up a can of worms.

TimRudy, 2024-01-19:
Similarly, thinking of improving the validations and errors:

Why doesn't the `iverilog` layer that does the "Verify" (Ctrl R) do a simple check like that and complain about incomplete `if`s and `case`s? It must be because of the open-ended, very flexible interpretation of the HDL. But does `iverilog` have some switch that would do a more strict check? That would make sense...

Would be brought to your attention early, long before sending to Yosys

cavearr, 2024-01-19:
I’m reviewing your sugestion. Give me some time to check it.

cavearr, 2024-12-21:
Hi @TimRudy @bogatkaj if you try the last wip https://dowloads.icestudio.io with the development toolchain (>=0.9.6) we are migrating to verilator instead of iverilog and i think this could detect this kind of deeper errors.

Try it please and tell me!

Plus, this wips has a lot of anoying ui errors fixed.

Thanks!

cavearr, 2025-01-09:
I'm closing the issue if is needed reopen it!

closed 2025-01-09 by cavearr