FPGAwars/icestudio #659: Running on M1 mac fails

AasmundN, 2022-11-08:
When trying to run the app, installed from here https://github.com/controversial/icestudio-m1, I get the error message: 

> “icestudio” is damaged and can’t be opened. You should move it to the Bin.

I am running macOS Monterey v12.2 on a MacBook Air (M1, 2020). I have Homebrew v3.6.5 installed and Python v3.9.

jrwagz, 2022-11-09:
I tried installing `0.10rc1` from this main repo on my MacBook Air (M2, 2022) and the app installs (albeit compiled for x86) and runs.  But sometimes it crashes, and some times the dropdown menus on the right side of the toolbar don't always display:

[image]

I haven't yet tried to actually compile and upload to a board, so it's not clear if those parts are working or not.

I was able to install the toolchain just fine, however when I try to install the drivers, I get this error:

[image]

AasmundN, 2022-11-09:
**Solution:**
Downloading the source code .zip file for the newest version from [here](https://github.com/FPGAwars/icestudio/releases) and making a few small changes to the code before running the program manually with ```npm run start```. The changes made were as follows:

1. Firstly the package **nw.js** could not be installed properly. The solution was as per  [this issue](https://github.com/FPGAwars/icestudio/issues/642) to add ```npm_config_nwjs_process_arch=x64``` to the command. I also had some issues with some other packages complaining, which I fixed with the flag ```--force```, which I doubt is good practice. Thus the command used to install the packages correctly was ```npm_config_nwjs_process_arch=x64 npm install --force```.

2. Secondly the installation path of the **homebrew** package manager is different for the Apple Silicon Macs than for the intel based Macs ([homebrew on m1](https://earthly.dev/blog/homebrew-on-m1/)). The icestudio source code uses homebrew to install some of its drivers from APIO, and assumes it is installed in the old, intel directory. Thus we need to replace all instances of ```/usr/local/bin/```with ```/opt/homebrew/bin/```, which you can do by simply using the replace all functionality in your IDE.

After making these changes run the app with ```npm run start``` and everything should work.

**Note** that when trying to verify, build or uploading the project to the board the operating system will complain about the app trying to use packages that are installed externally. We therefore need to allow the use of these packages as described [here](https://support.apple.com/en-us/HT202491). You will have to do this over and over until all the packages have been accepted, both for verify, build and upload.

This solution was tested with **Monterey v12.2** on a **Macbook Air (M1 2020)**.

crfosse, 2022-11-09:
> I tried installing `0.10rc1` from this main repo on my MacBook Air (M2, 2022) and the app installs (albeit compiled for x86) and runs. But sometimes it crashes, and some times the dropdown menus on [...]

The drop-down menus on the right side have moved to the menu on the lefthand sidebar. Press the blue box icon and you'll have access to the usual blocks.

jrwagz, 2022-11-09:
> **Solution:** Downloading the source code .zip file for the newest version from [here](https://github.com/FPGAwars/icestudio/releases) and making a few small changes to the code before running the [...]

Fantastic work!  I've made a patch and pull request based on these instructions.

AasmundN, 2022-11-09:
> Fantastic work! I've made a patch and pull request based on these instructions.

Great! Note that if you change the file paths so that they work with M1 Macs then the code probably would not work for intel based Macs.

jrwagz, 2022-11-09:
> Note that if you change the file paths so that they work with M1 Macs then the code probably would not work for intel based Macs.

Agreed.  This is why I simply added more paths, instead of changing the existing paths, so that it will work on both intel based macs and apple silicon macs.

AasmundN, 2022-11-09:
> > Note that if you change the file paths so that they work with M1 Macs then the code probably would not work for intel based Macs.
> 
> Agreed. This is why I simply added more paths, instead of changing the existing paths, so that it will work on both intel based macs and apple silicon macs.

Great! I checked out the pull request and saw that you had considered it, hence the deleted comment.

jrwagz, 2022-11-09:
So, while this fix works if I run via `npm run start` from the source code, it doesn't work if I install the packaged and published version of the app:

[image]

Any ideas on what the difference is?

AasmundN, 2022-11-09:
I'm guessing that although the fix has been merged with the current development branch, the fix still hasn't made it into the published version. If you want a packaged version you can use ```npm run buildOSX``` to generate the .dmg file.

jrwagz, 2022-11-12:
Couple of items

1) I was indeed testing with the most recent version of the WIP builds.  The version (as shown in my screenshot above) is `0.9.2w202211090311` and matches the latest WIP build version.

1a) I notice that there are multiple possible `.dmg` downloads on the WIP site, is there a difference between them that is relevant?
[image]

2) Creating a local dmg build natively on apple silicon
> I'm guessing that although the fix has been merged with the current development branch, the fix still hasn't made it into the published version. If you want a packaged version you can use `npm run buildOSX` to generate the .dmg file.

I attempted to run `npm run buildOSX` on my machine, but it fails to run as shown below.  I'm testing on **MacOS Ventura 13.0**  I do see a 13.0.1 update is available, perhaps I'll try installing that and see if it helps.

```
$ npm run buildOSX

> icestudio@0.9.2w buildOSX
> grunt dist --platform=darwin

Executing Gruntfile.js...

------------ INFORMATION FOR DEBUGING -------------------
* Package name: icestudio-0.9.2w202211110711
* NW Version: 0.60.0
* APPIMAGE: dist/icestudio-0.9.2w202211110711-linux64.AppImage
* DMGIMAGE: dist/icestudio-0.9.2w202211110711-osx64.dmg
* Target platform: osx64
* SubTASK for the DIST task:
┌─────────┬─────────────────────┐
│ (index) │       Values        │
├─────────┼─────────────────────┤
│    0    │      'jshint'       │
│    1    │    'clean:dist'     │
│    2    │ 'nggettext_compile' │
│    3    │     'copy:dist'     │
│    4    │    'json-minify'    │
│    5    │       'nwjs'        │
│    6    │  'exec:repairOSX'   │
│    7    │  'compress:osx64'   │
│    8    │      'appdmg'       │
│    9    │     'clean:tmp'     │
└─────────┴─────────────────────┘
---------------------------------------------------------
Running "jshint:all" (jshint) task
>> 29 files lint free.

Running "clean:dist" (clean) task
>> 0 paths cleaned.

Running "nggettext_compile:all" (nggettext_compile) task

Running "copy:dist" (copy) task
Created 1000 directories, copied 12048 files

Running "json-minify:json" (json-minify) task

Total compressed: 209 files
>> 573.093 KiB - 6.9% = 533.443 KiB

Running "json-minify:ice" (json-minify) task

Total compressed: 56 files
>> 593.765 KiB - 47.4% = 312.03 KiB

Running "nwjs:src" (nwjs) task
Using v0.60.0 (sdk)
Create cache folder in /Users/justinwagner/projects/icestudio/cache/0.60.0-sdk
Downloading: http://dl.nwjs.io/v0.60.0/nwjs-sdk-v0.60.0-osx-x64.zip
  downloading [====================] 100% 0.0s

Create release folder in /Users/justinwagner/projects/icestudio/dist/icestudio/osx64
>> NW.js app created.

Running "exec:repairOSX" (exec) task
Cleaning OSX bundle
    Check dist/icestudio/osx64/icestudio.app/Contents/Frameworks/nwjs Framework.framework/Helpers
    =>Preserving dist/icestudio/osx64/icestudio.app/Contents/Frameworks/nwjs Framework.framework/Helpers
    Check dist/icestudio/osx64/icestudio.app/Contents/Frameworks/nwjs Framework.framework/Internet Plug-Ins
    =>Preserving dist/icestudio/osx64/icestudio.app/Contents/Frameworks/nwjs Framework.framework/Internet Plug-Ins
    Check dist/icestudio/osx64/icestudio.app/Contents/Frameworks/nwjs Framework.framework/Libraries
    =>Preserving dist/icestudio/osx64/icestudio.app/Contents/Frameworks/nwjs Framework.framework/Libraries
    Check dist/icestudio/osx64/icestudio.app/Contents/Frameworks/nwjs Framework.framework/Resources
    =>Preserving dist/icestudio/osx64/icestudio.app/Contents/Frameworks/nwjs Framework.framework/Resources
    Check dist/icestudio/osx64/icestudio.app/Contents/Frameworks/nwjs Framework.framework/Versions
    =>Preserving dist/icestudio/osx64/icestudio.app/Contents/Frameworks/nwjs Framework.framework/Versions
    Check dist/icestudio/osx64/icestudio.app/Contents/Frameworks/nwjs Framework.framework/nwjs Framework
    =>Preserving dist/icestudio/osx64/icestudio.app/Contents/Frameworks/nwjs Framework.framework/nwjs Framework
Repairing OSX bundle
>> cp: dist/icestudio/osx64/icestudio.app/Contents/Frameworks/nwjs Framework.framework/Versions/A/*: No such file or directory
>> cp: dist/icestudio/osx64/icestudio.app/Contents/Frameworks/nwjs Framework.framework/Versions/A: No such file or directory
    =>OSX bundle OK

Running "compress:osx64" (compress) task
Warning: unable to stat srcFile (dist/icestudio/osx64/icestudio.app/Contents/Frameworks/nwjs Framework.framework/Helpers) Use --force to continue.

Aborted due to warnings.
```

3) For the record, if I use `npm run start` locally, I can still install the drivers as expected, and even flash an UPDuino 3.1 board, so I think it can work, we just need to figure out what is going wrong in the build processes.

jrwagz, 2022-11-12:
So, I upgraded to **MacOS Ventura 13.0.1** and re-tested, and all behavior remains the same.

However, I did learn something new.  I found out that even though I am unable to install the drivers using the WIP build, the WIP build does allow me to upload to a device, assuming that you've previously installed the drivers with the app running from `npm run start`.

So, from that perspective, this should be "functional" for me know, but this will still be a hurdle for new users on Apple Silicon, as it looks like at this point those users will have to build locally to get the drivers installed.

cavearr, 2022-11-12:
Hi! build in osx M1 or M2 is a little tricky, i have in my todo list document well the process, i'll try to do this weekend to help you and that you help us to debug . Between today and tomorrow, i'm improving the build.

Thanks a lot for your support @jrwagz , i'm maintaining thenews in this thread

cavearr, 2022-11-13:
i'm working in the m1 native build, but i need more time, there are some problems with the nwjs engine for M1 and i need to  develop and try some things.

I'm maintaining you inform in this thread the next days.