FPGAwars/icestudio #665: Add support to Colorlight 5A-75E with FT232RL JTAG programmer

roby2014, 2022-12-19:
Hello.
I have a Colorlight 5A-75E board (v8.0) however, looks like IceStudio does not support to program it via `FT232RL`.

I was able to program it via command line following the documentation: https://trabucayre.github.io/openFPGALoader/guide/advanced.html#ft231-ft232-bitbang-mode-and-pins-configuration

Would it be possible to add support for this JTAG programmer on IceStudio?

benitoss, 2022-12-28:
In theory all JTAG programmers supported by OpenFPGALoader could workk with ICEStudio to program the Colorlight FPGA boards.
However, the design of ICEStudio doesn't allow to work with different JTAG programmers for the same board.
So, One board has associated one JTAG Programmer, and it is necessary to duplicate the board in ICEStudio to work with different JTAG programmer. That is the reason that you can see for example un the ECP5 board parts, the same Colorlight board with a different JTAG programmer in parenthesis.
It is not complicated to create a new copy of the Colorlight 5A-75E board (v8.0) with FT232RL as a JTAG programmer but in the same way, other user can ask support for several JTAG programmer for each Colorlight board. It should be crazy.

So, the best solution requires a complete modification of the ICEStudio and Apio to support for the same board different JTAG programmer.

Anyway, I don't recommend to use the FT232RL in bitbang because it is too slow programming the board.
The recommended JTAG programmer for all Colorlight FPGA boards is the FT2232H programmer because it is not too expensive,  it is really fast and it is supported in ICEStudio for all Colorlight FPGA boards.
You can buy one here:
https://es.aliexpress.com/item/32806818411.html?spm=a2g0s.12269583.0.0.5add6f4bElJ1Vx&gatewayAdapt=glo2esp

roby2014, 2023-01-12:
> Anyway, I don't recommend to use the FT232RL in bitbang because it is too slow programming the board.

I understand that is not "recommended", but not everyone has the same hardware/money. IMO, there should be as maximum JTAG programmers for each board as possible so people don't need to waste 20€ on a JTAG programmer like the FT2232H (instead of 1€ on a FT232RL).

benitoss, 2023-01-12:
For $8 including shipping you can get the FT232H programmer here:
https://www.aliexpress.us/item/2251832861120673.html

And for less than $4-$5 including shipping you can get the USB Blaster Programmer here:
https://www.aliexpress.us/item/3256801662617764.html

I don't think that $5 or $8 is too much for a JTAG programmer that is considerably faster than FT232RL

Of course that the FT232RL can be used as a programmer in IceStudio because it is compatible with openFPGALoader, but the question is if it is valuable for the community.

benitoss, 2023-01-21:
Let me send you  examples of the speed when you program the board with different programmers. 
My tests are done in the EBAZ4205 Xilinx board but I think the results are similar with the Colorlight Boards.

FT2232H programmer (similar to FT232H) -->  1.75 seconds
USB-Blaster programmer --> 15.76 seconds
STM32 (Blue Pill) --> 194.72 seconds

The EBAZ4205 board has 35K Logic Elements and the Colorlight board has 25K, so the time to program the Colorlight will spend almost 1/3 less time.
STM32 works similar to the FT232R. As you see it spends a minimal of 2 minutes for program a board with FT232R, when the USB-Blaster spends ~11 seconds.  
You can see the attached files with the results

[image]
[image]
[image]
[image]
[image]
[image]

Regards
Fernando

roby2014, 2023-01-23:
Yes, **I do understand that those programmers are waaaaaay faster than the FT232RL**!
That's not the point here, it was just a suggestion because I would prefer sticking up with this programmer instead of buying another one. The point of this PR is to add support to hardware that some people may have available.

cavearr, 2025-01-19:
Hi! @benitoss had been do a great work around this boards, today the last wip https://downloads.icestudio.io contains the last colorlight boards added by @benitoss  that try to put the best way to program it.

I'm closing the issue and if you need reopen it.

closed 2025-01-19 by cavearr