FPGAwars/icestudio #669: Array Input/Output Blocks

Byter64, 2023-01-22:
I have started using icestudio and I am building a simple adder in order to get into icestudio.

As of now I can't seem to find an arry input so that I can for example get my first summand as one block with say 32 outputs (or maybe even just one that represents all 32) instead of 32 blocks with one output. 

So my question is: Does there anything exist like an array input or array output block?

Obijuan, 2023-01-22:
Yes. Please, check the iceK collection documentation. There you will find some examples

https://github.com/FPGAwars/iceK/wiki

For example, if you want to enter a 4-bit number from 4 external switches, you have to name the input with the "[3:0]" suffix, like button[3:0]

The same for the outputs

Obijuan, 2023-01-22:
In this example a 2-bits contants is shown on the LEDs:

[image]

You will find more example on the icek collection wiki

closed 2023-01-22 by Obijuan

Byter64, 2023-01-29:
[image]

I already stumbled across this feature but it doesn't work when you uncheck the "FPGA pin" option. That's what confuses me. (See picture: Test1, has the option checked, Test2 not)

If this decision is intentional I find it a bit irritating from user perspective

Obijuan, 2023-01-29:
In your example, Test1 is an 8-bits input port in the Top level design (main circuit). It should always be connected to physical pins in order to work on real hardware (when working directly with verilog, you need a constraint file. But in Icestudio you just indicate the physical pins in this block)
On the contrary, Test2 is an 8-bits input port for a module. It does not correspond to any phycal pins. It is used to connect the internal modules with other, but inside the top level (no access to outside)

So, if you uncheck the FPGA pin checkbox in TEST1, you convert it to a generic input port, and lose the associated pins. But this is something you usually do not have to do, because in the top level design all the ports should be connected to any physical pin (always)

As an example, if you want to create a circuit for adding, let's say, two 4-bits numbers coming from outside switches, and showing the result if 4 external LEDs, the top level design should be something like this:

[image]

Inside the Adder block, you will find the two input ports and one output port. These ports are not connected to any physical pin. This allows you to use this module in any other design or FPGA

[image]

I encourage you to have a look at the examples of the icek collection, and play with them to learn the basics. Then check the icewires (for accessing buses), iceGates and iceMuxes collections

You can find some implemetantion of adders in the iceArith collection, although this collection  is not documented yet

If you have more questions, please, feel free to ask them in the FPGAwar group  (You can ask in English)

I will keep this issue closed