FPGAwars/icestudio #675: Toolchain installation issues

muxia1954, 2023-02-06:
The used PC was broken
Install a new PC(windows 11) for icestudio
 However,the tool chain cannot be installed with the error.
Can't use it in windows11?

An error occurs, generates a Create virtualenv...20%, 
and does not continue from

--- Error message below ---
error executting command pyton.exe,m
venv,"C:\user\admin\icestudio\venv"

help me

cavearr, 2023-02-06:
What python version have you installed?

muxia1954, 2023-02-06:
Ver.3.9 in the installation package.

muxia1954, 2023-02-07:
I installed the toolchain again this morning
Installation without error

Thank you very much、Muchas gracias 
ありがとう。

cavearr, 2023-05-19:
Download the last wip at https://icestudio.io. i think this issue is solved. If you try it and works, plase tell us in this thread.

cavearr, 2023-06-29:
For the moment i close this issue, if problem persists open it again.

closed 2023-06-29 by cavearr