FPGAwars/icestudio #677: Upduino 3.1 No Basic > Input Menus after toolchain install

mccoyg, 2023-02-15:
I am on a Mac and these menus are missing after board selection and stable toolchain install.

[image]

all menus missing even though successfully installed toolchain.

mccoyg, 2023-02-15:
Its supposed to look like this:
[image]

jojo535275, 2023-02-16:
Hello
This is normal, to simplify the logic, about 1 year ago, we converted the Basic menu to a floating "toolbox" removing it from the toolbar.
To access to this toolbox, you have two options :
- go to menu Edit / toolbox
- Just hit Ctrl +T on your keyboard (or corresponding keys for mac)
For the remaining menus that you mentioned, they were made to pick-up blocks in a selected collection. 
Now you have to use the collection manager to pick up all your blocks in the collections ( previously loaded in your icestudio).

Obijuan, 2023-02-16:
In this animation you can see the new approach:
[image]

Inside the toolbox you will see the default collection and all the collections that you have installed manually

mccoyg, 2023-02-16:
All ok, thank you for your quick response to the newb!

closed 2023-02-16 by mccoyg