FPGAwars/icestudio #682: Icestudio has problems with Python 11

HttpRafa, 2023-02-26:
I think IceStudio doesn't understand Python version 3.11.2. 
Because I get an error during the toolchain installation that at least 3.7 must be installed.

[image]
[image]

gfcwfzkm, 2023-04-22:
Just installed the program myself and I get the exactly same error. Got python 3.11 installed as well.

cavearr, 2023-05-19:
What operating system are you using?

HttpRafa, 2023-05-22:
At this time i was using Windows 11

bruno504luis, 2023-05-28:
Any advice by now? I've got win8.1 and same problem.

bruno504luis, 2023-05-28:
> 

Ok, I used vers 3.10.11 it worked.

Chandler-Kluser, 2023-06-06:
I am using Endeavour OS and I have had the same issue in installation:

[image]

[image]

I believe there is an issue with python 3.11

cavearr, 2023-06-06:
Hi @Chandler-Kluser !, could you execute :

- which python
- which python3
- python --version
- python3 --version

Chandler-Kluser, 2023-06-06:
> Hi @Chandler-Kluser !, could you execute :
> 
>     * which python
> 
>     * which python3
> 
>     * python --version
> 
>     * python3 --version

[image]

cavearr, 2023-06-29:
Hi! could you install the last WIP version from https://downloads.icestudio.io and try if it works on your environment?

GhostDog98, 2023-06-29:
> Hi! could you install the last WIP version from https://downloads.icestudio.io and try if it works on your environment?

@cavearr I've been having this same issue with the same results for `which python`, this fixed the issue, perfect timing as I installed icestudio not 5 minutes ago :laughing:

cavearr, 2023-06-29:
Thanks!!! if is it necessary reopen the issue! :)

closed 2023-06-29 by cavearr

Chandler-Kluser, 2023-07-01:
It worked!!! Thanks a lot!!!