FPGAwars/icestudio #685: Installation stuck on "Creating venv" at 20% when installing toolchain

ziggekatten, 2023-03-15:
OS: Pop_OS!

Tried both tarball and AppImage, same issue. icestudio.log:
```
=======================================================================================
 Icestudio session Tue Mar 14 2023 17:04:34 GMT+0100 (Central European Standard Time)
 Version: 0.10.1w202303080503
=======================================================================================
Node information:
  * Node version: v19.3.0
  * lts: undefined
  * SourceURL: https://nodejs.org/download/release/v19.3.0/node-v19.3.0.tar.gz

NW information: 
  * NW version: 0.72.0
  * NW-flavor: sdk
  * Chromium: 109.0.5414.87

System information:
  * Architecture: x64
  * Platform: linux

Profile file: /home/mats/.icestudio/profile.json

- PROFILE:

{
  "data": {
    "board": "upduino31",
    "boardRules": true,
    "collection": "/tmp/.mount_omQxOl/usr/bin/resources/collection",
    "language": "en",
    "uiTheme": "dark",
    "externalCollections": "",
    "externalPlugins": "",
    "remoteHostname": "",
    "showFPGAResources": false,
    "displayVersionInfoWindow": "yes",
    "lastVersionReview": "0.10.1w202303080503",
    "loggingEnabled": false,
    "loggingFile": "",
    "pythonEnv": {
      "python": "",
      "pip": ""
    }
  }
}

- PATHs

common.BASE_DIR: Icestudio base dir: /home/mats
common.ICESTUDIO_DIR: Icestudio folder: /home/mats/.icestudio
common.PROFILE_PATH: Profile path: /home/mats/.icestudio/profile.json
common.INTERNAL_COLLECTIONS_DIR: Internal collections: /home/mats/.icestudio/collections
common.APIO_HOME_DIR: APIO folder: /home/mats/.icestudio/apio
common.ENV_DIR: Python virtual environment: /home/mats/.icestudio/venv
common.ENV_BIN_DIR: Executable files: /home/mats/.icestudio/venv/bin
common.ENV_PIP: PIP executable:  /home/mats/.icestudio/venv/bin/pip3
common.APIO_CMD: Apio command: APIO_HOME_DIR="/home/mats/.icestudio/apio" PATH="/home/mats/.icestudio/venv/bin:/tmp/.mount_NhmUmk/usr/bin:/tmp/.mount_NhmUmk/usr/sbin:/opt/m68k/bin:/opt/m68k/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" "/home/mats/.icestudio/venv/bin/apio"
Common.APP: Icestudio APP folder: /tmp/.mount_NhmUmk/usr/bin
common.APP_DIR: Icestudio execution folder: /tmp/.mount_NhmUmk/usr/bin

===> tools.CHECKTOOLCHAIN
>>>> utils.executeCommand => APIO_HOME_DIR="/home/mats/.icestudio/apio" PATH="/home/mats/.icestudio/venv/bin:/tmp/.mount_NhmUmk/usr/bin:/tmp/.mount_NhmUmk/usr/sbin:/opt/m68k/bin:/opt/m68k/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" "/home/mats/.icestudio/venv/bin/apio" --version

>>(ERROR): /bin/sh: 1: /home/mats/.icestudio/venv/bin/apio: not found

----!!!! ERROR !!!! -----
CMD: APIO_HOME_DIR="/home/mats/.icestudio/apio" PATH="/home/mats/.icestudio/venv/bin:/tmp/.mount_NhmUmk/usr/bin:/tmp/.mount_NhmUmk/usr/sbin:/opt/m68k/bin:/opt/m68k/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" "/home/mats/.icestudio/venv/bin/apio",--version
  Error flag: true
  Error: true
------> MENU ENTRY POINT: Install Toolchain
**** STEP: Check internet connection
**** STEP: Check Python
**** STEP: Create virtualenv
>>>> utils.executeCommand => /usr/bin/python3 -m venv "/home/mats/.icestudio/venv"
```

Obijuan, 2023-03-17:
Could you please open a terminal an execute the following command on your system?:

```
/usr/bin/python3 -m venv "/home/mats/.icestudio/venv"
```

I need to check what's going on

ziggekatten, 2023-03-17:
Running the command creates the venv. However, starting icestudio, it still hangs at the same spot. Ended up installing the toolchain manually as per documentation. So it seems to be the call to python from the app to create venv works, but the tolchain stuff does not (apio etc)

percz, 2023-04-02:
I just had to copy apio in...
_cp venv/bin/apio .icestudio/venv/bin/_

I'm running Mac M1 aka Darwin-arm64, but otherwise same icestudio.log

cavearr, 2023-04-02:
Are you running with rosetta?

TurBoss, 2023-04-30:
hello,
I managed to get it running by installing apio from the venv

```
source ~/.icestudio/venv/bin/activate
pip install --upgrade pip
pip install --upgrade apio
```

after that run icestudio

hope it works there :)

cavearr, 2023-04-30:
Thanks for your solution!!!

closed 2023-04-30 by cavearr

Chandler-Kluser, 2023-06-10:
> Could you please open a terminal an execute the following command on your system?:
> 
> ```
> /usr/bin/python3 -m venv "/home/mats/.icestudio/venv"
> ```
> 
> I need to check what's going on

Thanks a lot!! This command helped me to solve a similar issue!!