FPGAwars/icestudio #686: problem uploading to upduino 3.1

itsme200, 2023-03-18:
So i bought an upduino 3.1 for a school project. I installed Ice studio and selected the board. I checked that the version was 0.10.0 since only this one supports it. At first I was getting the ftdi_usb_get_strings failed: -4 (libusb_open() failed) error. after that I installed apio manually since that could of have been an issue. And after that the apio system --lsftdi outputed that one device was found and the name was single rs232. So that was sorted. I enabled the drivers and after trying to upload again it only outputed the Upduino 3.1 not connected error. I have tried multiple fixes but none of them worked. Any help would be apreciated.
[image]
[image]

vr2045, 2023-05-17:
Looks like the FTDI EEPROM got blown away somehow so you're seeing this as a 'single rs232'. This should show up as a UPduino 3.0/3.1 so icestudio can recognize it.
Please use the script [here ](https://github.com/tinyvision-ai-inc/UPduino-v3.0/tree/master/Board/EEPROM) to reinitialize the EEPROM then unplug/replug the board. Should then get recognized by icestudio.

dee2dee3, 2023-12-25:
im having a similar problem where i cant upload to my upduino because Icestudio cant seem to recognize it, would this solution work for me? if so how do I do it

cavearr, 2024-02-05:
Hi !! yo fix the problem? i'll keep open the issue some more days awaiting for your response and if not i'm closing it.

The problem is more related to upduido than icestudio.

Reopen when you want if you encountered it  closed.

vr2045, 2024-02-06:
Hello,
Please stop by the tinyVision.ai discord server so we can help you:
https://discord.gg/3qbXujE
Agreed, this is not an icestudio issue and should be closed.

closed 2024-02-06 by cavearr