FPGAwars/icestudio #687: Bidireconal Switch

maxall41, 2023-04-06:
Apologies if this is the wrong place to post this I'm pretty new to icestudio and FPGAs in general.

Im trying to create a bi-directional switch on an I2C data line but I can't seem to get it to build no matter what I do I always get the error:
```
libc++abi: terminating with uncaught exception of type nextpnr_ice40::assertion_failure: Assertion failure: dst != WireId() (/work/_builds/darwin-x64/nextpnr-ice40/nextpnr/ice40/delay.cc:160)
```
whenever I try to build my code. I have attached the icestudio file and a screenshot of my design is below. Does anybody know why this is happening?
[image]

[basic_switch.ice.zip](https://github.com/FPGAwars/icestudio/files/11166899/basic_switch.ice.zip)

cavearr, 2023-05-26:
What is your board?

cavearr, 2025-01-24:
I'm closing this issue for no feedback received. No in the last wip https://downloads.icestudio.io you could use inout ports that do easier the use of tristate ports.

If you need something more, reopen it.

closed 2025-01-24 by cavearr