FPGAwars/icestudio #688: Installed Collections don't show up

byteit101, 2023-05-23:
I have freshly installed IceStudio 0.10, and have added several collections (IceK, etc), but they don't show up.

It took me a long time to find the default collection, as it's not intuitive, and not documented anywhere that I could find other than the release notes. However, I can't seem  to get anything to show up beyond the default collection. Selecting a collection is highly confusing, and doesn't seem to actually do anything. Closing the default collection and clicking the blue lego button again only pops the default collection back up. Restarting icestudio only shows the default collection.

How can I use the collections I just installed?

cavearr, 2023-05-26:
Hi!, first put all the collections in the same folder (choose a location in your disk to use as collection folder).

Then go to Edit->Preferences->External Collections 

[image]

And put in the input field the absolute path to your collection folder.

Then close and reopen Icestudio and click at collection manager icon (blue lego block).

You need to wait some time while iceStudio scan your disk and folders, we are improvement this panel to do more intuitive, but now if you wait some minutes (depends the number of collections and the speed of your disk) the collections appear.

I'm recommend you work with the last wip, you can download from https://icestudio.io. 

Tell us if all works!

byteit101, 2023-05-26:
> You need to wait some time while iceStudio scan your disk and folders, [...] if you wait some minutes (depends the number of collections and the speed of your disk) the collections appear.

This appears to be the answer, as I have left ice studio open in the background for the past 3 days since I reported this, and the collections have appeared with no work on my part. I didn't need to change the external collections folder. This is very unintuitive.

cavearr, 2023-05-26:
three days? have you installed the last wip? what operating system to you use?

byteit101, 2023-05-26:
I have installed 0.10. Debian 11 x64. `/` on SSD, `/home` on HDD

Granted, I haven't checked it in 3 days. After futzing about for 5-10 minutes, I gave up, minimized it, and hadn't checked it or un-minimized it until 20 minutes ago.

cavearr, 2023-05-26:
Install the last wip, because 0.1 has some bugs around the collection manager. Anyway i'm working in it to do more intuitive. 

Thanks for your feedback! 

Enviados: Sábado, 27 de Mayo 2023 0:21:18

cavearr, 2023-05-26:
I'm close this issue, if you have any news or problems, reopen it.

thanks again!

closed 2023-05-26 by cavearr