FPGAwars/icestudio #689: Wrong clock pin rule for iCESugar-Pro boards?

GhostDog98, 2023-06-29:
According the icestudio, pin P3 is the clk input pin:
[image]

However, according to the [docs](https://github.com/wuxx/icesugar-pro#clock) the clock is connected to pin P6 
[image]
This is confirmed by the schematic file:
[image]

I may be misreading this, but it appears to be incorrect.

GhostDog98, 2023-06-29:
I'm also unsure if this is why icestudio seems to never successfully be able to make code that runs on my board...

cavearr, 2023-06-30:
@benitoss @jojo535275 , could you check if there is something wrong with this pin of icesugar pro?

benitoss, 2023-06-30:
You right, the correct Pin is P6.
I have already created a pull request with the Fix
https://github.com/FPGAwars/icestudio/pull/690

cavearr, 2023-06-30:
Thanks a lot @benitoss !, @GhostDog98 there is a new WIP a https://downloads.icestudio.io download it and check if all works!

tell us please!

GhostDog98, 2023-06-30:
Yes it now says P6, but even the simplest examples don't seem to work still. I'm not sure if this is an issue with icestudio or myself, but uploading the .bit file compiled by icestudio (pictured below) using icesprog fails to produce any signal on A8, but if i use the official makefiles and modify the icesugar pro "blink" script to use the same pin A8, it does output a signal for A8. This makes me think that either the makefile that icestudio is using, or the toolchain is out of date or not configured correctly... I'll investigate more if I can but I'm not too familiar with icestudio's internals. 

icestudio in question:
[image]

benitoss, 2023-06-30:
@GhostDog98 In the case of ECP5 boards the option of export the bit file doesn't work.
If you don't use the FT2232H , FT232H or USB blaster JTAG programmers, you have to take the bit file by hand in the directory where the source code is generated internally and program the board with the icesprog application.

Anyway, the best solution is generating a new board configuration with icesprog as the way to program the board. 
For it, could you tell me which is the command line for programming the board with the icesprog application?
Thanks

GhostDog98, 2023-06-30:
Hi, I'm not sure I understand what you are saying about getting the source file internally and programming with icesprog, as I believe that's what I'm doing. I go into the project folder, then into /ice-build/Untitled/ and use icesprog on `hardware.bit`, but this still fails.

As for the command line tool I am using, I simply built it from source from [here](https://github.com/FPGAwars/toolchain-icesprog.git), but iCESugar offers its own binary, which seems nearly identical that also works for me. ([here](https://github.com/wuxx/icesugar/tree/master/tools))

benitoss, 2023-06-30:
@GhostDog98 when you program the board with hardware.bit file  typing iceprog hardware.bit you are writing the SPI Flash and it spends a lot of time.
There is other option with iceprog application to write in the SRAM of the FPGA board that is almost instantaneous. Check the options of the iceprog and test the same hardware.bit with the new option.
Later tell me what conmand line you used and I will add the new mode of iceprog to the IceSugar Pro board in Icestudio

GhostDog98, 2023-06-30:
Apologies for the delay, as I have never really gotten deep into programming tools like this. I can't for the life of me figure out what option in icesprog allows me to write to the SRAM. Additionally, while using dapprog/openocd to write the files *does* make things faster (only about 25 seconds), it does not make the files compiled by icestudio *work*, so I see little point in adding another upload tool if the files that icestudio generates don't work yet for this board... I'm honestly very lost on what to do at this stage... :sweat_smile: 

Let me know if there's anything specific I can do to help. When I wake up tomorrow, I'll see if I've left a jtag adapter in any of my storage bins.

Thanks for the continued help :heart:

GhostDog98, 2023-07-02:
Hi All, 

I've managed to identify a workaround, and subsequently the issue, but have been unable as of yet to fix it within icestudio due to my lack of familiarity in the programs git repo structure...

My workaround:
Copy over make file from known working project (i chose the blink project from the example repo, available [here](https://github.com/wuxx/icesugar-pro/blob/master/src/blink/Makefile))
run `make clean` to remove icestudios messed up build
rename lpf, v, and json files to be common name
Redefine TARGET and TOP as well as OBJS to that prior common name

Now we should have:
-rw-r--r-- 1 ghostdog ghostdog 699445 Jul  2 12:47 hardware.json
-rw-r--r-- 1 ghostdog ghostdog    182 Jul  2 12:47 hardware.lpf
-rw-r--r-- 1 ghostdog ghostdog    629 Jul  2 12:47 hardware.v
-rw-r--r-- 1 ghostdog ghostdog    503 Jul  2 12:50 Makefile

Run make
Upload the program using icesprog or dapprog/openocd.
Verify output on pin A8 using a multimeter as HIGH (3v3).

This shows that the issue is in whatever `make` style system icestudio uses, rather than the programmer or the vhdl that is generated. Where can I find the make files that icestudio uses? I've looked [here](https://github.com/FPGAwars/icestudio/blob/develop/app/scripts/services/compiler.js) but couldn't find it, and can't think of anywhere else to look...

GhostDog98, 2023-07-02:
Follow up: Installing apio development version through the icestudio menu fixes builds, but switching back to the "stable" version makes the stable version work now?!?!? I'm going to lose what's left of my sanity at this rate...

benitoss, 2023-07-02:
@GhostDog98 , Icestudio is the application to generate the structure of your Verilog code. Therefore in Icestudio you select first the board (usually asociated with a programmer in case of several ECP5 boards)
Appio is the application used for creating the synthesized bit file and programming the board.
The programming of the board is done called another programmer program. 
For example there are boards that use dapprog or iceprog, that is not the issue.
The problem is that for program your board in SRAM using iceprog, this application needs to work with the SVF file format that is NOT directly generated by Appio.
That is the reason that I recommend to use a external JTAG programmer. It will program the board really fast in SRAM.
If you insist in using iceprog application we need to program the SPI Flash which spends more time. 
Please send me the command line for iceprog but you have to use the BIT file version for program the board.
Regards

cavearr, 2024-02-06:
Hi! if there is no feedback in some days, i'll close this issue. If you need to reopen it later, do it!

Thanks!

cavearr, 2024-03-18:
I'm closing the issue for not having reply.

If you need help about it , reopen it please!

closed 2024-03-18 by cavearr