FPGAwars/icestudio #691: Mac Toolbar Missing Items

dan-sherwin, 2023-07-10:
I just got Icestudio installed on my Mac.  Im using an Alchitry Cu iCE40-HX8K board.  I did a successful install of app and toolchain.  I am able to load any of the example projects, build it, upload it to the board, and it works as expected.  

The problem is, the IDE toolbar has none of the basic items I expected to see like the dropdowns like Bit, Logic, and Setup.  The only things I have is File, Edit, View, Select, Tools, Basic, and Help.  Im attaching a screen shot.  

[image]

Obijuan, 2023-07-10:
Please, have a look at issue #677

The Icestudio blocks are available inside the new toolbox window

dan-sherwin, 2023-07-10:
I looked for duplicate issues, I swear, but I missed finding that one.  Thank you and sorry for the dup issue.

closed 2023-07-10 by dan-sherwin

Obijuan, 2023-07-10:
Hey! No problem 😀️ . It is usually difficult to find similar issues because Icestudio is still in development and it is constantly changing...