FPGAwars/icestudio #693: how to run icestudio on 64bit raspbian linux on raspberry pi

mvdhoning, 2023-07-22:
Are there step by step instructions on how to get this running on a raspberry pi. There are nice step by step instructions for windows and x86/x64 linux but not for arm linux.
There is a 64bit arm linux build that actualy is a 32bit arm linux build and does not work on 64bit arm. #639 
Regarding raspberry pi there was also issue #426 in the closing comment is refered to list of builds but those appear also to be 32bit and not 64bit.
So next i tried following the wiki on running the development version of icestudio but that ends up in error on the 
npm install --legacy-peer-deps 
step.
```
1331 verbose stack Error: command failed
1331 verbose stack     at ChildProcess.<anonymous> (/home/user/.config/nvm/versions/node/v19.3.0/lib/node_modules/npm/node_modules/@npmcli/promise-spawn/lib/index.js:53:27)
1331 verbose stack     at ChildProcess.emit (node:events:513:28)
1331 verbose stack     at maybeClose (node:internal/child_process:1098:16)
1331 verbose stack     at ChildProcess._handle.onexit (node:internal/child_process:304:5)
1332 verbose pkgid nw@0.60.0
1333 verbose cwd /home/user/fpga/icestudio
1334 verbose Linux 6.1.21-v8+
1335 verbose node v19.3.0
1336 verbose npm  v9.2.0
1337 error code 1
1338 error path /home/user/fpga/icestudio/node_modules/nw
1339 error command failed
1340 error command sh -c node scripts/install.js
1341 error Response code 404 (Not Found)
1342 verbose exit 1
````
Unfortunately this does not give me any hints on what goes wrong. Other then that is missing/cannot find a file.
Looking in script folder there is a preInstallArm64.sh running that from the icestudio directory and next running npm install --legacy-peer-deps seemingly gives the same file not found error.
So before i go install 32bit raspbian linux, what steps can i do next to get icestudio running on 64bit arm linux.

benhaotang, 2023-08-30:
Any update on this? I also failed to install on my jetson orin nano, I'm not getting some error but completely freezed. Have tried the pre-compiled arm64 zip, up to 0.9.1 can run without problem, but any version after 0.10 all returns: bash: ./icestudio No such file or directory, I have properly set up the permission.

benhaotang, 2023-09-07:
I think this is due to the nw compatibility with arm device, see here https://github.com/nwjs/npm-installer/issues/83, don't know if this can be easily solved

benhaotang, 2023-09-07:
tried the prefix here https://github.com/FPGAwars/icestudio/commit/69cdd1f57ee427f5567529ebdd9fdaeee202ff6f, on both my wsl on windows on arm and ubuntu aarch64 on mw jetson nano, I all just get 
[image]

I'll try recompile nw on my machine today to see if it can fix, I guess https://github.com/FPGAwars/icestudio/issues/639 this is also due to nw having incorrect arm builds

cavearr, 2024-02-05:
Hi! i'm releasing a new wip for linux arm 64 native, you can download from https://downloads.icestudio.io 

This is a zip archive that you only need to execute icestudio executable file to run it

benhaotang, 2024-02-07:
> Hi! i'm releasing a new wip for linux arm 64 native, you can download from https://downloads.icestudio.io
> 
> This is a zip archive that you only need to execute icestudio executable file to run it

Hi, I can run and compile without problem on my jetson with ubuntu 20.04 arm, Thank you a lot for your help! the new version is also a lot snappier.

cavearr, 2024-02-07:
Thanks for your feedback @benhaotang ! in the next weeks i'm releasing new things very usefuls i hope you enjoy it!

cavearr, 2024-03-18:
I'm closing the issue, i think with the new WIPS , icestudio runs well in raspberry pi 4

If you need help about it , reopen it please!

closed 2024-03-18 by cavearr