FPGAwars/icestudio #694: Icestudio wants to install Python 3.9.9 while I already have Python 3.11.3 installed.

wgaylord, 2023-07-28:
I am trying to install the Windows 64-bit version of icestudio on Windows 10. (Version 0.10 according to the file) And the install insists on installing Python 3.9.9. 

I have tried the nightly / WIP build `0.10.1w202306300406` which is the newest one and it to wants to install Python 3.9.9 for some reason. 

Running `where python` turns the following on my system. 

```
C:\Users\chibill>where python
C:\Python311\python.exe
C:\Users\chibill\AppData\Local\Microsoft\WindowsApps\python.exe
```
Where the weird path in AppData is Microsoft's stupid thing to try and make you install python thru the appstore is its not installed on your system. 

Any solutions would be greatly appreciated.

cavearr, 2023-07-29:
Hi! i'm checking it and updating window installer, tell you soon.

wgaylord, 2023-08-15:
Any information? Just tried the nightly / WIP build `0.10.1w202307290607` and no change.

cavearr, 2023-08-15:
Hi! i'm working in a new installer, should be ready along tihs week, but if you can't wait, for the moment you could download de zip version, from https://downloads.icestudio.io/  decompress an execute icestudio directly.

[image]

cavearr, 2024-02-08:
the last wip works with python 12 , when you try to install icestudio, the python installer detetcts your previous python and ask you for install or not, donwload an try it if you have any problem reopen this.

Thanks!

closed 2024-02-08 by cavearr