FPGAwars/icestudio #696: icestudio runs pushbutton example for OK-iCE40Pro failed on MacOS

nandemoi, 2023-08-20:
Example:
Basic → 06. Pushbutton and LED.ice

Error Message:
yosys -p "synth_ice40 -json hardware.json" -q main.v
nextpnr-ice40 --up5k --package sg48 --json hardware.json --asc hardware.asc --pcf main.pcf -q
scons: *** [hardware.asc] Error -11

OS:
Ventura 13.4 (intel Core i3)
failed too on Ventura 13.4.1 (Apple M1)

By the way the flattened MacOSaarch64 (ARM64) didn't seem to contain an MacOS execution file.

Thank you.

cavearr, 2023-08-20:
hi! Do you install apio toolchain? click on "install development version"

[image]

And if it is possible, install the latest wip from https://downloads.icestudio.io

nandemoi, 2023-08-21:
Yes, I did. The LED only examples load fine.

cavearr, 2023-10-09:
How we don't have notices about it, I'm closing the issue, if you try the possible solution commented or have any other comment, reopen.

Thanks!

closed 2023-10-09 by cavearr