FPGAwars/icestudio #698: Incompatible library  version

EnriqueGP22, 2023-09-16:
Hola a todos, me compré una Alhambra II en julio y conseguí instalar Icestudio en un MacBock Air Ventura 13.4.1(c) (aunque sólo pude la versión 0.10.1w202306300406, las demás versiones daban errores o no encontraban los drivers). Bueno, con este mac y la Alhambra II he disfrutado mucho de las enseñanzas y ejemplos de Obijuan, pero la pantalla es muy pequeña. Tengo un Mac de27" más viejo, con High Sierra 10.13.6 en el que he conseguido instalar, con la ayudas del forum, los drivers y el Icestudio, aunque en este sólo pude instalar la versión 0.9.0-1 (es la única que reconoce los drives), y  APIO 8.2 (he probado con 8.0, y 8.1). Carga los ejemplos perfectamente,  pero a la hora de sintetizar o cargar da el siguiente error:
"Incompatible library version
libmtl.8.dylib requires version 9.0.0 o later, but libconv.2.dylib provides version 7.0.0.. scons: ***
Hardware error 6."
Os agradecería mucho que me ayudarais aunque ya que debe ser maravilloso disfrutar del Icestudio y varias ventanas con 27". 
Gracias anticipadas. Enrique
Adjunto el log
[icestudio.log](https://github.com/FPGAwars/icestudio/files/12640321/icestudio.log)

cavearr, 2023-09-17:
Hola Enrique, ahora mismo, en la última versión que usamos en el motor nwjs no soportan osx Sierra, ya que está basado en chrome y chrome dejó de dar soporte para esa versión de OSX.

En pocos meses vamos a cambiar algunas cosas importantes en el motor y eso va a ayudar a qeu Icestudio pueda funcionar en casi cualquier equipo, viejo y nuevo, estoy haciendo foco en ello pero aún nos queda algo de tiempo.

Una cosa que se me ocurre que puedes hacer por el momento, mientras tanto, es instalarte ubuntu en un virtualbox y ahí ejecutar la última versión de icestudio.

Siento no poder ayudarte  más ahora mismo, aunque te digo en poco tiempo ya podrás utilizarlo sin problemas.

EnriqueGP22, 2023-09-19:
Hola Carlos,

Muchas gracias por la respuesta y por seguir trabajando en para que el Icestudio funcione con equipos viejos, que siguen funcionando aunque no existan para el fabricante. Intentaré cuando tenga un rato, instalar el ubuntu en una virtualbox y ver que pasa, seguro que funciona. Muchas gracias por la sugerencia.
Te contesto aquí porque esta cerrado el hilo.
Saludos
Enrique

cavearr, 2023-09-19:
Claro Enrique! perdona por cerrar el hilo, lo vuelvo a abrir y así cuando haya novedades sobre este tema actualizo aquí.

cavearr, 2026-07-01:
Current stable version is outdate, we are in process now to release the new one (current branch) if you want to try it download from https://downloads.icestudio.io

I'm closing this issue because is outdated and we need to reduce noise for real bugs in the new pre-release, if you needed reopen it.