FPGAwars/icestudio #702: Permissions issues

Bexin3, 2023-10-23:
Hello, for some reason every single script says its from an unidentified developer and I am having to allow permissions one by one which is taking really long.

Bexin3, 2023-10-23:
(I am on macOS, M1 Pro)

Bexin3, 2023-10-23:
Further, it seems like it can't be use at the same time as game porting toolkit as game porting toolkit requires the installation to be in usr

cavearr, 2023-10-23:
Hi @Bexin3 ! i'm working in including a fix for this in the toolkit installation. This happens because the tools are compiled with open source tools without apple propietary licenses.

for the moment to solve it without trust by hand all of the items, open a terminal and execute:

cd .icestudio
find . -exec xattr -d com.apple.quarantine {} \;

The first command go into  your .icestudio folder that it is in your home folder and inside it icestudio install all of things that it need.

The second command  trust the libraries and executables .

Execute after you install the toolchain or while install it and appear the first message.

Tell me if it works

juanuicich, 2023-12-17:
I had the same issue and the following command fixed it under macOS Sonoma 14.2 using zsh 5.9 as shell.

```
sudo find . -print0 | xargs -0 xattr -d com.apple.quarantine
```

Thanks @cavearr for the hint.

Habraken, 2024-01-08:
I seem to have a similar issue. Neither of the suggestions above worked for me. I am on an Intel Mac running Sonoma 14.0.
[image]

cavearr, 2024-02-05:
Hello friends! You already have the official Icestudio package for native ARM64, I have also improved the installation process (signed application for OSX M1 and goodbye to headaches with permissions).

When installing the toolchains, when the system requires it, it will request the administrator password to modify the permissions.

You can download it from the github actions or more conveniently from https://downloads.icestudio.io

Thanks @ProbabilityEngineer for the initial feedback, it has been crucial.

I will keep the issue open for a few days in case you give me feedback and if you do so later and find it closed, open it without hesitation.

ProbabilityEngineer, 2024-02-07:
sudo xattr -cr /Applications/icestudio.app  fixed it for me. This is a very broad command.

you could also try
sudo xattr -r -d com.apple.quarantine /Applications/icestudio.app

and

sudo codesign --force --sign - /Applications/icestudio.app

ProbabilityEngineer, 2024-02-07:
Another issue is that installing the dev toolchain 0.9.1 installs the intel package. I'm not sure but thought the previous build installed the apple silicon package (once I'd swapped out nwjs in icestudio.app). Maybe I installed it manually I don't remember.

cavearr, 2024-02-07:
Hi @ProbabilityEngineer! today i'm review it, are you sure you are executing from arm terminal? the  dev toolchain detects your architecture and install the correct one , maybe you are running a terminal with roseta?, the m1 dmg is signed now  with the command similar than you write. but not publish it in gruntfile for not publish my certs and for many people than don't have xcode installed.

I'll test today and tell you 

The permissions problems should be fixed , a prompt for

ProbabilityEngineer, 2024-02-07:
Hi there,
Using icestudio-0.11.3w202402071202-osxarm64.dmg
I installed the dev toolchain from the menu option in Icestudio. It installed the intel version of 0.9.1.
My terminal is not running in rosetta.

I tried installing this twice on a system without ~/.icestudio or ~/library/application support/icestudio , downloading the dmg separately each time.

If I try to open icestudio-0.11.3w202402071202-osxarm64.dmg with gatekeeper enabled then gatekeeper refuses to open it.

If I disable gatekeeper with sudo spctl --global-disable then the dmg will open but I get the error "icestudio.app" is damaged and can't be opened. You should move it to the Trash.

If I repair icestudio.app with
sudo xattr -r -d com.apple.quarantine /Applications/icestudio.app
(I have left gatekeeper disabled)
It will run and will install the toolchain and ask for my password when it does so. 0.8.4 installs and simulates, builds and uploads without triggering gatekeeper (gatekeeper still disabled).

When I select "install development version" within Icestudio it asks for my password and installs the toolchain, it installs apio 0.9.1, but the executables in /Users/sam/.icestudio/apio/packages/tools-oss-cad-suite/libexec are intel, not apple silicon.

********************************
What happens with icestudio-0.11.3w202402040902-osxarm64.dmg

I have to disable gatekeeper to open the dmg
Running icestudio.app results in ""icestudio.app" is damaged and can't be opened. You should move it to the Trash"

After I repair it with xattr -r -d com.apple.quarantine /Applications/icestudio.app
it runs

cavearr, 2024-02-07:
Thanks a lot for this detailed test!!! I’m working on it una fee hours Thanks again!

ProbabilityEngineer, 2024-02-07:
Observations:
Try installing on a Mac account not signed in to your developer account and see how it behaves.

The 2 ways I can think of avoiding the extended attributes error (xattr) 
1: using a package installer with admin privileges an ad-hoc signature.
2: Signing the app with a paid apple developer account.

ProbabilityEngineer, 2024-02-07:
[image]

cavearr, 2024-02-07:
Thanks for the tips the dmg is being signed with paid Apple account , the problem could be in appdmg package , i’m on it

ProbabilityEngineer, 2024-02-07:
I know you're working on your method, this is my latest brute-force installer package, along with the packages project file and the packages app and the post install script.

https://www.dropbox.com/scl/fi/tmmj3ornavfege9xjv0t8/2024-02-07-installer-with-toolchain.tgz?rlkey=2eofzbulcg2hx6mcnhz999hol&dl=0

cavearr, 2024-02-07:
Thanks @ProbabilityEngineer ! i know this is not the best method, but this very clear and we have deployed for the automatic compilations and bundle generation against PRs, it's very important that it not be "tricky" as much as we could. 

But all of your ideas , feedback and efforts are very useful for me, thanks a lot again.

zapta, 2024-02-07:
I am trying to install for M2 Sonoma 14.3 and am getting permission errors, after I select a board.

What is the best practice at this point?

cavearr, 2024-02-07:
Hi team! in short i'm generating a new release that i think it works well, give me little time!

zapta, 2024-02-07:
This just worked for me with the stable release.

sudo xattr -cr /Applications/icestudio.app

cavearr, 2024-02-07:
Yes, for the stable release this is the solution tos fix the osx problems.

I'm recomending you install the latest wip from https://downloads.icestudio.io, our wips are very stable and have the latest yosys toolchains and a lot of functionalities.

This afternoon a new wip will be launched with the osx arm 64 native support fixed, i'm telling you  soon

ProbabilityEngineer, 2024-02-07:
I really appreciate this thank you.

ProbabilityEngineer, 2024-02-08:
I was just able to test this and it works exactly as described. YAY!

cavearr, 2024-02-08:
I'm going to close the issue, if any other problem appears around this, reopen!

ProbabilityEngineer, 2024-02-09:
I was wrong. downloads.icestudio.io listed the arm64 versions as ".error"? so I downloaded and installed the intel version.
When I realized today I downloaded the arm64 version. 

It installs and runs. it installs the 0.8.4 toolchain.

When you update to the dev toolchain it installs the venv, downloads tools-oss-cad-suite-darwin_arm64-0.1.0.tar.gz, throws an error and doesn't unpack the bin, lib or libexec folders.

Those folders are present in the tar.gz file.

I don't know if this is just my system or what. It was a clean install.

[image]

cavearr, 2024-02-10:
Hi @ProbabilityEngineer we are generating new toolchain versins this days and there is a bug in the last one. During the day i'm fixing it.

Thanks for your patience this "jump" has been hard!

cavearr, 2024-02-10:
Hi again! @ProbabilityEngineer try to download again the dmg from https://downloads.icestudio.io
Make sure you donwload 0.11.3w202402070902 version, if you don't see it, refresh browser, the bug is relationship with github api that cache the request and returns invalid artifacts.

If you download this version, all should be ok.

I'm working in a new download and update system ,that will be ready  very soon.

ProbabilityEngineer, 2024-02-10:
Hi there!
0.11.3w202402070902 Is the version that I was reporting on that failed to unpack the dev toolchain. I will try it again.

I don’t want you to feel rushed, I appreciate your hard work. I hope to provide you with useful information along the way.

Thanks!

cavearr, 2024-02-10:
don't worry @ProbabilityEngineer , all your feedback is very very welcome! try again because this .error message could be for the github api error , in the new "update manager" i'll pretend to not use github as download repository because this problems appears sometimes and it is weird.

Thanks again and report all of you find!

ProbabilityEngineer, 2024-02-10:
I tried again with icestudio-0.11.3w202402070902-osxarm64.dmg . It worked!

From a clean system (I deleted the app, ~/.icestudio, and /application support/icestudio),

The app installed, toolchain 0.8.4 installed, then the dev toolchain installed. It verifies, builds and uploads on my nandland go board.

The only thing I did differently was that I used the 0.8.4 toolchain to verify and build before I updated to the dev toolchain.

Thanks!

cavearr, 2024-02-10:
Yeah!!!! thanks a lot for your feedback!!