FPGAwars/icestudio #705: IceStudio release Aarch64 actually is a Aarch32 build.

janrinze, 2023-11-14:
The AArch64 release apparently was built as a Aarch32 executable.
Please build as Aarch64 as it won't run on 'newer' systems that don't support Aarch32 such as Apple M1,M2.. (Asahi Linux.)

```
janrinze@debian:~/Desktop/icestudio-0.11-aarch64$ file ./icestudio 
./icestudio: ELF 32-bit LSB pie executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.2.0, BuildID[xxHash]=b16e95ba333ec97c, stripped

janrinze@debian:~/Desktop/icestudio-0.11-aarch64$ ./icestudio 
bash: ./icestudio: cannot execute binary file: Exec format error

janrinze@debian:~/Desktop/icestudio-0.11-aarch64$ uname -a
Linux debian 6.5.0-asahi-00671-g618f14cf48b9 #2 SMP Thu Oct 12 22:10:35 CEST 2023 aarch64 GNU/Linux
```

cavearr, 2023-11-14:
Hi @janrinze, I'm working on it, very soon we have e nice new version with aarch64 well supported. I'm maintaining this thread to tell you about the progress.

janrinze, 2023-11-14:
@cavearr Thanks for the quick reply! will keep an eye out here.

Sam99C, 2024-01-21:
If you just want a native apple silicon version you can re-wrap it with nwjs-osx-arm84 by replacing Icestudio.app "macos" and "frameworks" folders with those from nwjs-v0.83.0-osx-arm64 nwjs.app . Works for me.

cavearr, 2024-01-21:
Hi sam! Thanks a lot  for the tip! I’m trying it.

Sam99C, 2024-01-22:
I discovered that you should repair the app's permissions afterwards with 

chmod -R 755 /Applications/icestudio-new.app

otherwise it's broken for anyone but the user who replaced the nw.js files

cavearr, 2024-02-05:
Hello! I'm releasing new wips with native support for arm 64 for linux and osx, you can download them from github action or more easily from https://downloads.icestudio.io

If you try it, please tell me if everything works well. In the case of OSX arm64 I have improved the installation process to avoid headaches with permissions and annoying things, I will continue working on the installation process of the next wips for all operating systems and I would greatly appreciate your comments.

I will keep the issue open for a few days waiting for your responses. If I close it and you later want to add something, reopen it without hesitation.

ProbabilityEngineer, 2024-02-07:
sudo xattr -cr /Applications/icestudio.app

fixed it for me. This is a very broad command.

you could also try:
sudo xattr -r -d com.apple.quarantine /Applications/icestudio.app

and:
sudo codesign --force --sign - /Applications/icestudio.app

ProbabilityEngineer, 2024-02-07:
Another issue is that installing the dev toolchain 0.9.1 installs the intel package. I'm not sure but thought the previous build installed the apple silicon package (once I'd swapped out nwjs in icestudio.app). Maybe I installed it manually I don't remember.

cavearr, 2024-02-08:
I am closing this issue because with the latest wip this malfunction is solved :)

If you find any problem related to this, please reopen it.

closed 2024-02-08 by cavearr