FPGAwars/icestudio #706: Bundle required Python

tmssngr, 2023-11-15:
I'm using the Windows `.zip` bundle. It can't find a Python. Why don't you bundle the required version?

cavearr, 2023-11-15:
What version of icestudio are you using? what version of python do you have installed?

tmssngr, 2023-11-16:
I have unpacked the latest version from the release-section here. Why should I install any Python? Why can't it be part of the distributed `.zip` file?

cavearr, 2023-11-16:
Hi! i'll recomend you the latest WIP: 

[ https://downloads.icestudio.io/ | https://downloads.icestudio.io ] 

And download the MSI installer (this install python with Icestudio). 

Check it and tell me if it works. 

Enviados: Jueves, 16 de Noviembre 2023 7:46:34

cavearr, 2023-11-16:
Hi! i don't understand whats happen! i'll try the url, what version of windows do you have? what is your web browser and version? if you pobided me, will be very useful for me to fix a bug probably.

In other way, try with google chrome please, you should view something like this, and download the .exe (MSI) bundle:

[image]

tmssngr, 2023-11-16:
Sorry, it was a problem of uMatrix not refreshing correctly the page. I could download it now, but it tries to **install** an outdated Python. Why not bundle it (in the meaning of copying) as subdirectory of the installation? I hate it if applications a) require external dependencies or b) install something else that remains after uninstalling the application, or c) a different application wants to install a different version of Python. It's completely fine if you bundle an older Python version that is known to work flawlessly with icestudio.

PS: BTW, I was able to try icestudio on Linux today and like it.

cavearr, 2023-11-16:
Very soon we don't bundle python anymore, but  this is the python in the bundle now, this windows bundle is thinked for students of one univerity that needed and this is a version compatible with other things that they need.

Anyway you could uninstall this version of python and install the latest from https://www.python.org/

I'm happy that you install in linux :).  if you have more questions reopen this thread, for the moment i'm closing it! 

Thanks for your feedback!

closed 2023-11-16 by cavearr