FPGAwars/icestudio #708: ERROR:cert_verify_proc_builtin.cc(621)] No net_fetcher for performing AIA chasing.

jiapei100, 2023-12-07:
Failed to build **icestudio** under **Ubuntu 22.04.3**. 
Well, I upgraded all packages to the newest, but I don't think that will influence **cert_verify**???
Does it have something to do with **SSL** version?
Maybe it's again a **SSL version 1 vs. version 3** issue?

```console
➜  icestudio git:(develop) ✗ grunt serve              
Executing Gruntfile.js...

------------ INFORMATION FOR DEBUGING -------------------
* Package name: icestudio-0.11.2w202312070212
* NW Version: ^0.82.0
* APPIMAGE: dist/icestudio-0.11.2w202312070212-linux64.AppImage
* DMGIMAGE: dist/icestudio-0.11.2w202312070212-osx64.dmg
* Target platform: linux64
* SubTASK for the DIST task:
┌─────────┬─────────────────────┐
│ (index) │       Values        │
├─────────┼─────────────────────┤
│    0    │      'jshint'       │
│    1    │    'clean:dist'     │
│    2    │ 'nggettext_compile' │
│    3    │     'copy:dist'     │
│    4    │    'json-minify'    │
│    5    │       'nwjs'        │
│    6    │ 'compress:linux64'  │
│    7    │ 'appimage:linux64'  │
│    8    │     'clean:tmp'     │
└─────────┴─────────────────────┘
---------------------------------------------------------
Running "nggettext_compile:all" (nggettext_compile) task

Running "watch:scripts" (watch) task
Waiting...
Executing Gruntfile.js...

------------ INFORMATION FOR DEBUGING -------------------
* Package name: icestudio-0.11.2w202312070212
* NW Version: ^0.82.0
* APPIMAGE: dist/icestudio-0.11.2w202312070212-linux64.AppImage
* DMGIMAGE: dist/icestudio-0.11.2w202312070212-osx64.dmg
* Target platform: linux64
* SubTASK for the DIST task:
┌─────────┬─────────────────────┐
│ (index) │       Values        │
├─────────┼─────────────────────┤
│    0    │      'jshint'       │
│    1    │    'clean:dist'     │
│    2    │ 'nggettext_compile' │
│    3    │     'copy:dist'     │
│    4    │    'json-minify'    │
│    5    │       'nwjs'        │
│    6    │ 'compress:linux64'  │
│    7    │ 'appimage:linux64'  │
│    8    │     'clean:tmp'     │
└─────────┴─────────────────────┘
---------------------------------------------------------
Running "exec:stopNW" (exec) task

Running "exec:nw" (exec) task
>> [2026026:2026026:1207/022949.718054:ERROR:policy_logger.cc(157)] :components/enterprise/browser/controller/chrome_browser_cloud_management_controller.cc(163) Cloud management controller initialization aborted as CBCM is not enabled.
>> [1207/022949.759408:ERROR:elf_dynamic_array_reader.h(64)] tag not found
>> [2026061:2026074:1207/022954.768502:ERROR:cert_verify_proc_builtin.cc(621)] No net_fetcher for performing AIA chasing.
>> [2026061:2026074:1207/023004.007883:ERROR:cert_verify_proc_builtin.cc(621)] No net_fetcher for performing AIA chasing.

Done.
Completed in 257.887s at Thu Dec 07 2023 02:34:03 GMT-0800 (Pacific Standard Time) - Waiting...
```

cavearr, 2023-12-26:
Hi @jiapei100 ! sorry for the delay in my response, i'll try to reproduce your three issues and tell you soon.

cavearr, 2024-02-05:
Hi you are using and old release. We are working in a kind of rolling relases like a wip versions (work in progress) that you could download from github actions and more comfortable from https://downloads.icestudio.io

Please download the last one and try it. I'm closing this issue if the problem persists with the last wip reopen it!

Thanks!

closed 2024-02-05 by cavearr