FPGAwars/icestudio #709: Fatal error: EINVAL: invalid argument, open 'app/resources/collection/blocks/Logic/Combinational/Mux 2:1.ice'

jiapei100, 2023-12-07:
And, it looks **2:1** is **NOT** accepted under **Ubuntu 22.04.3** ? 
And, it seems **Mux 2:1.ice** is a temparary file generated ? 
I think that file name **Mux 2:1.ice** is **NOT** accepted by my operating system **Ubuntu 22.04.3**, wondering how to standardize it? 

```console
Running "unzip:using-router" (unzip) task
Fatal error: EINVAL: invalid argument, open 'app/resources/collection/blocks/Logic/Combinational/Mux 2:1.ice'
npm ERR! code 3
npm ERR! path ....../icestudio
npm ERR! command failed
npm ERR! command sh -c grunt getcollection && npmpd && cd app && npm install

npm ERR! A complete log of this run can be found in: ~/.npm/_logs/2023-12-07T10_50_48_271Z-debug-0.log
```

cavearr, 2024-02-05:
plase explain your steps or how are you doint to try to help you.

cavearr, 2024-03-18:
I'm closing the issue for not having reply and not be a global problem detected in other users.

If you need help about it , reopen it please!

closed 2024-03-18 by cavearr