FPGAwars/icestudio #713: noob questions

jmio23, 2024-01-10:
I got an IceFun HX8K, the one with the LED matrix on it, but I can't figure out which board I need to choose in IceStudio to get started? Had issues building the IceFun examples (gave errors when hitting build, but also the program converted the examples for IceFun, which probably complicates matters for me..)
 

Perhaps there is a better board for me to get started with? The IceFun examples i tried wouldn't build, and I'm too much of an FPGA noob to even know where to start figuring out why or what difference there might be for my specific board...

Also how do I know if the board is actually connected to IceStudio? can't see any visual indication, although windows detected/installed and connects fine apparently

jojo535275, 2024-01-10:
Hello jmio,

i do not have this board, but, Normaly, for the icefun board (HX8K), you have to select:
[image]

Also have  you installed the "Apio" toolchain? :
[image]

The easiest way to know if you card is well connected, is to try loading a simple blinking led design on your board.
[image]
 - you should accept the alhambra to icefun conversion of the design (in the pop-up window)
 -  and then you have to choose the led you want to blink (in the drop-down menu of the output port(in blue on the above picture  )

cavearr, 2024-02-05:
I'm closing this issue because i think is solved.

If problem persist of you need to continue it, reopen!

Thanks!

closed 2024-02-05 by cavearr